bblanchon/ArduinoJson #333: Problems with large Json

pbecchi, 2016-08-16:


pbecchi, 2016-08-16:
I am trying to decode a large Json file: tipically a Weather Underground Api response:
 the buffer that contain the file is about 2500 bytes!
I am using an ESP8266 with Arduino Core libraries.
I have routine:
void JsonDecode(char json[],String names[],float values[]);
that does the decoiding and put results in values.
If I try with: 
StaticJsonBuffer<any size> jsonBuffer;
i got a crash!
if I use:
DynamicJsonBuffer jsonBuffer;
i got a successfull decoding but after a fe loop the program crash....again!

Thanks for your help

bblanchon, 2016-08-16:
`StaticJsonBuffer` made the device crash because it was too big.

Parsing fails after a few iterations because you're reusing the `DynamicJsonBuffer`.

See:

* [FAQ: Why does my device crash or reboot?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-does-my-device-crash-or-reboot)
* [FAQ: The first parsing succeeds, why do the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-parsing-succeeds-why-do-the-next-ones-fail)

pbecchi, 2016-08-16:
Thanks for your fast replay!

WHY TOO BIG: there should not be memory limitations using an ESP8266 ( I have a  free memory of about 25k!)
Any limitation in size coming from your library ? Then...What is the max size for StaticJsonBuffer?

I don't reuse the DynamicJsonBuffer but I am afraid there there could be some memory leak (very common using Heap memory).

If there is a defined memory limit , for a input stream of >2000 bytes I should probably use another tool .

bblanchon, 2016-08-17:
The maximum size of the stack is defined by the compiler.
If you're using Arduino to program your ESP8266, the size of the stack is 4KB.

No, there is no size limitation imposed by the library, and I seriously doubt there is a memory leak.
Please provide a **short** code sample that reproduces the issue.
 
See:

* https://github.com/esp8266/Arduino/issues/808
* [FAQ: I found a memory leak in the library!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#i-found-a-memory-leak-in-the-library)

pbecchi, 2016-08-18:
Thanks again for your answer: the issue #808 you have referenced show a case very similar to mine:
unfortunately it doesn't show  any the solution!!
If the stack MAX is 4k (this is really a bad surprise!!) I cant have a buffer of 2500 bytes (this is the size of the json the wunderground.com provide) and an additional StaticJsonBuffer!

Then I have to use a DynamicJsonBuffer ..... but in that case I have:
-or a crash when I read from the web
-or a return without reading anything
-or a crash in a random location (stream operation) may be after a few successful Json decoding
(I check and the free Heap is bigger than 25k!)

Other solution could be to allocate the StaticJsonBuffer<2000> globally....but that will mean to reuse it several times!

This is an extract of my code (I used cut&paste so it could be not complete!)

`byte prova() {
	WiFiClient client;
	const int httpPort = 80;
	if (!client.connect("api.wunderground.com", 80))
	{
		client.stop();
		Serial.println("connection failed ");
		return 0;
	}
	String url = "GET ";
	String streamId = "/api/wukeycode/conditions/q/Italy/pws:ISAVONAL1.json";

	url = url + streamId
		+ " HTTP/1.1\r\n" + "Host: " + "api.wunderground.com" + "\r\n" + "Connection: close \r\n\r\n";
	client.print(url);
	ulong millismax = millis() + 10000;
	while (!client.available() && millis() < millismax)delay(10);
	if (!client.available())SPL("no rep");
	char json[2500]; int i = 0;
	while (client.available()&&i<2500){
		json[i++] = client.read();
	}
	json[i] = 0;
	client.stop();
	DynamicJsonBuffer jsonBuffer;
				JsonObject& root = jsonBuffer.parseObject(json);
				Serial.print("m.a.h."); Serial.println(ESP.getFreeHeap());
				// Test if parsing succeeds.
				if (!root.success()) {
					SPL("Weather parseObject() failed");
					return 0;
				}
				else {
					SPL("Weather Parsing...");
					for (byte i = 1; i < Nval+1 ; i++) {
						if (nomi[i] == "local_epoch")
						{
							time_t time = root["local_epoch"];
							val[i - 1] = (float)time;
						}
						else
						{
							float valore = root[nomi[i]];		
							val[i - 1] = valore;
						}
					}
					return 1;
				}
}`

bblanchon, 2016-08-19:
You need to move the variable `json` to the heap.
The simplest solution is to use a `String` instead of a `char[]`, and to use `client.readString()`.

pbecchi, 2016-08-21:
Thanks

I have tried with String but it doesnt change: I still get "parseobjet failed"!

It look like there is no way to decode a 3000 bytes Json buffer!

bblanchon, 2016-08-21:
Please check the content of the `json` variable. I bet that you're not skipping the HTTP headers.

See:

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* https://github.com/bblanchon/ArduinoJson/issues/280#issuecomment-220632964

pbecchi, 2016-08-21:
No , no headers! I am cutting away first part of the HTTP response to reduce the size of the buffer to less than 2500 bytes.
May be that cutting It is creating some format problem for the decoder?
What are the essential rules for the correctness of the json buffer?
First character need to be a { ?

Thanks again for your help!

bblanchon, 2016-08-22:
Please copy/paste the content of `json` here.

pbecchi, 2016-08-22:
,	"current_observation": {
		"image": {
		"url":"http://icons.wxug.com/graphics/wu2/logo_130x80.png",
		"title":"Weather Underground",
		"link":"http://www.wunderground.com"
		},
		"display_location": {
		"full":"Loano, Italy",
		"city":"Loano",
		"state":"",
		"state_name":"Italy",
		"country":"IY",
		"country_iso3166":"IT",
		"zip":"00000",
		"magic":"17",
		"wmo":"16122",
		"latitude":"44.124748",
		"longitude":"8.254445",
		"elevation":"23.00000000"
		},
		"observation_location": {
		"full":"Loano Meteo di IW1PPB, Loano, Liguria",
		"city":"Loano Meteo di IW1PPB, Loano",
		"state":"Liguria",
		"country":"IT",
		"country_iso3166":"IT",
		"latitude":"44.124748",
		"longitude":"8.254445",
		"elevation":"98 ft"
		},
		"estimated": {
		},
		"station_id":"ISAVONAL1",
		"observation_time":"Last Updated on August 22, 2:04 PM CEST",
		"observation_time_rfc822":"Mon, 22 Aug 2016 14:04:50 +0200",
		"observation_epoch":"1471867490",
		"local_time_rfc822":"Mon, 22 Aug 2016 14:25:36 +0200",
		"local_epoch":"1471868736",
		"local_tz_short":"CEST",
		"local_tz_long":"Europe/Rome",
		"local_tz_offset":"+0200",
		"weather":"Clear",
		"temperature_string":"85.0 F (29.4 C)",
		"temp_f":85.0,
		"temp_c":29.4,
		"relative_humidity":"33%",
		"wind_string":"From the NNW at 10.0 MPH Gusting to 17.0 MPH",
		"wind_dir":"NNW",
		"wind_degrees":348,
		"wind_mph":10.0,
		"wind_gust_mph":"17.0",
		"wind_kph":16.1,
		"wind_gust_kph":"27.4",
		"pressure_mb":"1020",
		"pressure_in":"30.12",
		"pressure_trend":"-",
		"dewpoint_string":"53 F (12 C)",
		"dewpoint_f":53,
		"dewpoint_c":12,
		"heat_index_string":"83 F (28 C)",
		"heat_index_f":83,
		"heat_index_c":28,
		"windchill_string":"NA",
		"windchill_f":"NA",
		"windchill_c":"NA",
		"feelslike_string":"83 F (28 C)",
		"feelslike_f":"83",
		"feelslike_c":"28",
		"visibility_mi":"N/A",
		"visibility_km":"N/A",
		"solarradiation":"--",
		"UV":"-1","precip_1hr_string":"0.00 in ( 0 mm)",
		"precip_1hr_in":"0.00",
		"precip_1hr_metric":" 0",
		"precip_today_string":"0.00 in (0 mm)",
		"precip_today_in":"0.00",
		"precip_today_metric":"0",
		"icon":"clear",
		"icon_url":"http://icons.wxug.com/i/c/k/clear.gif",
		"forecast_url":"http://www.wunderground.com/global/stations/16122.html",
		"history_url":"http://www.wunderground.com/weatherstation/WXDailyHistory.asp?ID=ISAVONAL1",
		"ob_url":"http://www.wunderground.com/cgi-bin/findweather/getForecast?query=44.124748,8.254445",
		"nowcast":""
	}

bblanchon, 2016-08-22:
The beginning of the object is missing, isn't it?

pbecchi, 2016-08-22:
Yes , I had to cut it.... it was too large!

bblanchon, 2016-08-23:
Please realize that we need to be 100% sure that the JSON input is correct.

pbecchi, 2016-08-23:
Yes I understand, this is file extracted from the json WU output: the cutting point may be located few characters later  ... it matters?

bblanchon, 2016-08-24:
How can we validate the JSON input if you cut it?

pbecchi, 2016-08-24:
You must understand the full json is more than 3000 bytes now I cut it at about 2500 so I dont miss any data from it. If I dont cut it I will never be able to decode it!
There is any rule or any specific position where I have to cut it in order o be "validatet"?

bblanchon, 2016-08-25:
I'm not sure I understand what you're trying to do here.
ArduinoJson needs the complete JSON to parse it; it cannot parse incomplete or invalid JSON.
Why do you cut the JSON input?

jbrewkeggin, 2016-08-25:
Try validating your JSON with http://jsonlint.com.  The library will not parse your string unless that site says your string is valid json

pbecchi, 2016-08-25:
@bblanchon 
The reason is that the JSON is too large( that is the real problem and  the title of the ISSUE!).
We already discussed that if I don't cut it I will need a read buffer of about 3500 bytes plus other (?3 k) kbytes in the heap for the DynamicJsonBuffer: I don't have that memory available(6-7 kbytes).

Sorry but I don't really understand what means "validity" for that file, Anyhow ,in noway,I can handle the full file.
 
I am trying to decode it myself, just working with pointers in the reading buffer,to extract the data I need........

bblanchon, 2016-08-25:
By design, ArduinoJson requires that the whole JSON input is in memory, that is absolutely required by the zero-allocation/zero-copy strategy.

I advice you to use [JsonStreamingParser](https://github.com/squix78/json-streaming-parser) instead, just as project [esp8266-weather-station](https://github.com/squix78/esp8266-weather-station).

Sorry to see you leave.

closed 2016-08-25 by bblanchon

bblanchon, 2016-08-25:
This was added to the FAQ:

* [Can I parse a JSON input that is too big to fit in memory?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory)

bblanchon, 2016-08-25:
For the record, the JsonBuffer size should be about 1.5KB.

See:

* [FAQ: What are the common sizes for JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#what-are-the-common-sizes-for-jsonbuffer)