bblanchon/ArduinoJson #336: Q : Can the same JsonBuffer be used for reading (decoding )and writing (encoding) at the same time ?

vortex314, 2016-08-18:
I didn't find in the documentation ( wiki and examples ) if it is possible that : 
1/ You load a JsonBuffer with a string and parse/get the different elements
2/ You use the same JsonBuffer to set a number of fields.

The use case is the serialization of config to EEPROM/Flash. The idea was to store this as a string, load it in ArduinoJson , modify/read it, store back as string.

bblanchon, 2016-08-18:
Yes, it's possible, and it's a good practice 👍

closed 2016-08-18 by vortex314

bblanchon, 2016-08-18:
You're welcome Lieven, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

vortex314, 2016-08-18:
Benoit, just did the Github star. 
Adding the project on the page is out of my reach, don't master the mark-down syntax enough.
Project can be found here : https://sites.google.com/site/limero314/esp8266-to-stm32
A Wifi bootloader for STM32 chips.

bblanchon, 2016-08-19:
I added the link by I'm having trouble displaying the page, see the screenshot below:

[image]

vortex314, 2016-08-19:
Benoit, thanks again for the feedback, I changed the access rights of my document. It should work now.
I hope.

bblanchon, 2016-08-19:
Good. I can now see the images.