bblanchon/ArduinoJson #337: json string variable type error

interimlife, 2016-08-23:
I use ARDUINO IDE 1.6.11 from arduino.cc
I met errors in my program.
I can show these errors by JsonParserExample (included in library)

char json[] =  "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";  -->>original.
signed char json[] =  "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";  -->>error.
unsigned char json[] =  "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";  -->>error.
uint8_t json[] =  "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";  -->>error.
int8_t json[] =  "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";  -->>error.

change time key value.
char json[] =  "{\"sensor\":\"gps\",\"time\":135,\"data\":[48.756080,2.302038]}";

and change time variable. long to char.

unsigned char time = root["time"];  -> OK.
signed char time = root["time"];  -> OK.
uint8_t time = root["time"];  -> OK.
int8_t time = root["time"];  -> OK.
but..
char time = root["time"];  -> error.

I'm poor English. I beginning programming.
I don't know why I met these errors.

bblanchon, 2016-08-23:
`char`, `signed char` and `unsigned char` are three different types in C++.
A string literal is an array of `char`, not `signed char` or `unsigned char`.
That's why you have an error when changing the type of the variable `json`.

The second issue is a feature of ArduinoJson. 
The `char` type has been intentionally disabled to avoid an ambiguity.
Imagine you have this input `{'time':'8'}`, was should be the value of `root["time"]` as a `char`?
I see two possible answers. 
Either you see `char` as an integer; in that case, the result should be `8`.
Or you see `char` as a character; in that case, the result should be `56` (the ASCII code of character `8`)

I highly encourage you to use `char` to store characters, and `int8_t` or `uint8_t` to store integers.

closed 2016-08-30 by bblanchon

bblanchon, 2016-10-11:
Note: [this was added to the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-jsonvariant-cannot-be-converted-to-char)