bblanchon/ArduinoJson #338: JsonGeneratorExample.ino output different to your example

ControlFreakyKiwi, 2016-08-23:
Hello and thank you for your hard work and making your library available.
Please excuse my stupidity in advance, I am having trouble with your JsonGeneratorExample.ino example,

I have compiled your example exactly as you have written it, but the output on the serial port is below and is different to yours. I am using Windows 10, Arduino 1.6 and Arduino 2560 R3.

I would like to create a flat Json string (no nesting) with different data types, for some reason I cannot create a "key : signed long"  pair but a "key : double" pair works. (maybe you could show me a small example with a signed long)

I have tried reading all the FAQ and pitfalls , and after trying your unaltered example with unsuccessful serial output posted this question. thanks in advance.

Opening port
Port open
Opening port
Port open
{"sensor":[],"time":true,"data":8493}
{
  "sensor": [],
  "time": true,
  "data": 8493
}

ControlFreakyKiwi, 2016-08-24:
EDIT 2
I think this has something to with UNIX Windows line endings as I can now output the correct JSON string from the example in Visual Studio by changing to UNIX line endings then back to Windows line endings.

still having problems with signed long tho

`signed long sl = 1234567;
root["test signed long"] = sl; //this works outputs 1234567

struct cStepsFromZero

{
	signed long slAzimStepsFromZero;
	signed long slElevStepsFromZero;
};
cStepsFromZero mystepsfromzero;

root["Azimuth Steps From Zero"] = mystepsfromzero.slAzimStepsFromZero;// prints "true
Serial.println(mystepsfromzero.slAzimStepsFromZero); //prints the correct value
`

bblanchon, 2016-08-24:
Hi,

NOTE:
From there I'm going to assume that your first issue with Visual Micro is fixed.
If it's not the case, please open a **new** issue.
I'm going to change the name of the current thread to reflect the change in topic.

For your `signed long` that is serialized as `true`, I added a test case with the code you provided, but I cannot reproduce the bug.
Please see [Issue338.cpp](https://github.com/bblanchon/ArduinoJson/blob/7f5d7b7555ab189ac5eea227d0ff6903b89dff90/test/Issue338.cpp)
I've only tested on computers since I don't have access to an Arduino board right now, but there is no reason to get a different result.

I'm pretty sure you made a mistake or that the actual error is in a part of the code that is not visible here.

ControlFreakyKiwi, 2016-08-24:
Thanks for your time Benoît...............yes the Visual Studio /Visual Micro serial output works now fine by asserting the line endings.

You are probably correct  that I have made some sort of error (not really a programmer , more a cutter paster, meddler) as I seemed to have multiple problems at the same time however the lines below are sequential in the code and first line prints "true" and second line the correct signed long value.

Below is the actual code and the serial output 

Anyway thanks for your time, keep up the great work and I will continue banging my head :)

`void JSONgenerator()
{
	StaticJsonBuffer<600> jsonBuffer;

	JsonObject& root = jsonBuffer.createObject();
	root["Azimuth Angle"] = sunangles.dAzimuth;
	root["Elevation Angle"] = sunangles.dElevAngle;
	root["AzimuthStepsFromZero"] = mystepsfromzero.slAzimStepsFromZero;
	
	Serial.print("Arduino print ");
	Serial.println(mystepsfromzero.slAzimStepsFromZero);
	
	root.printTo(Serial);
	
	Serial.println();
}`

Arduino print 12271
{"Azimuth Angle":242.067,"Elevation Angle":-42.592,"AzimuthStepsFromZero":true}
Arduino print 12274
{"Azimuth Angle":242.018,"Elevation Angle":-42.626,"AzimuthStepsFromZero":true}
Arduino print 12277
{"Azimuth Angle":241.968,"Elevation Angle":-42.659,"AzimuthStepsFromZero":true}
Arduino print 12281
{"Azimuth Angle":241.919,"Elevation Angle":-42.692,"AzimuthStepsFromZero":true}

bblanchon, 2016-08-24:
The definition of mystepsfromzero is missing

ControlFreakyKiwi, 2016-08-24:
Hi Benoît,,

> The definition of mystepsfromzero is   missing

Thanks for looking, yeah the definition is there , I was trying to avoid repeating inserting code snippets
this is inside a file sunpos.h

`struct cStepsFromZero

{
    signed long slAzimStepsFromZero;
    signed long slElevStepsFromZero;
};

and this is `at the top of the ino sketch

`cStepsFromZero mystepsfromzero;

and the below prints the correct signed long value

`Serial.println(mystepsfromzero.slAzimStepsFromZero);

as a test I tried the below using "double_with_n_digits" which prints as a double (with 5 digits not 4)

void JSONgenerator()
{
	StaticJsonBuffer<600> jsonBuffer;

	JsonObject& root = jsonBuffer.createObject();
	root["Azimuth Angle"] = sunangles.dAzimuth;
	root["Elevation Angle"] = sunangles.dElevAngle;
	root["AzimuthStepsFromZeroDouble"] = double_with_n_digits(mystepsfromzero.slAzimStepsFromZero,4);
	root["AzimuthStepsFromZeroSignedLong"] = (mystepsfromzero.slAzimStepsFromZero);
	Serial.print("Arduino print ");
	Serial.println(mystepsfromzero.slAzimStepsFromZero);
	
	root.printTo(Serial);
	
	Serial.println();
}

serial output

Arduino print 2054
{"Azimuth Angle":46.867,"Elevation Angle":25.040,"AzimuthStepsFromZeroDouble":2054.00000,"AzimuthStepsFromZeroSignedLong":true}
`

bblanchon, 2016-08-25:
Also, are you using the latest version of ArduinoJson?

bblanchon, 2016-08-25:
[After more testing](https://github.com/bblanchon/ArduinoJson/commit/57b66d8dfe1fa3a4c737c95aa6a384a11474d70f), I'm pretty sure you're using an old version of ArduinoJson.
The latest version is 5.6.5.

You may need to flush the Arduino Library Manager cache, see instructions here:
https://github.com/bblanchon/ArduinoJson/wiki/Using%20the%20library%20with%20Arduino#for-arduino-16x-and-above

ControlFreakyKiwi, 2016-08-25:
Hi Benoît,
according to CHANGELOG.md , I have v5.6.5 but I will try flushing the library cache and get back and post the results.

ControlFreakyKiwi, 2016-08-25:
Upgrading to Arduino 1.6.11 has caused quite a few library out of scope errors for some reason that I will need to work my through  . I will post back

ControlFreakyKiwi, 2016-08-25:
Ok finally got to compile and upload with Arduino 1.6.11 with flushed library cache and confirmed ArduinoJson version is 5.6.5.

still printing"true" instead of signed long value

will try and see if there are any include files with inconsistent UNIX/Windows line endings

bblanchon, 2016-08-25:
There must be something wrong with the ArduinoJson version.
Indeed 5.6.5 would not print "2054.00000" but "2.0540e3", as in the [test case](https://github.com/bblanchon/ArduinoJson/blob/57b66d8dfe1fa3a4c737c95aa6a384a11474d70f/test/Issue338.cpp#L41).
Is it possible that you have another copy of the library that would be picked by the compiler for some strange reason.
Can you try to delete the library folder, then verify that the build fails as expected?

Is is possible to send a link to the complete file (possibly using pastebin)?

Can you try on another target, for instance, an Arduino UNO or Duemilanove?

ControlFreakyKiwi, 2016-08-25:
Well this is very interesting and confusing at the same time, I deleted the ArduinoJson folder from the libraries folder and tried to compile in Visual Studio and it failed as expected.

I deleted the file library_index.json from %LOCALAPPDATA%\Arduino15   (there was no library_index.json.tmp.gz to delete)

I also found an ArduinoJson.zip in %LOCALAPPDATA%\Arduino15\staging which I deleted

I then unzipped again ArduinoJson to my Arduino\libraries folder and recompiled and as you French say voila!. it works as expected.

Thank you again for your time , help and for your library, and especially the documentation , if English is your second language this is incredible work.

bblanchon, 2016-08-26:
It's "voilà", with an accent :stuck_out_tongue: 

Too bad I didn't think of this from the beginning, we would have saved time.
Anyway, you're very welcome, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-08-26 by bblanchon

ControlFreakyKiwi, 2016-08-26:
> It's "voilà", with an accent :stuck_out_tongue:
Yes good catch....we get a bit lazy out in the English colonies with our US keyboards......
Github star was added before I asked for help, pity more can't be added...
Thànks àgàin .... :stuck_out_tongue: