bblanchon/ArduinoJson #340: Passing NestedArrays to another function to populate with NestedObjects

treii28, 2016-08-26:
Two problems, in three questions.

First, I'm not a C++ guy, so I'm not sure if my first of two problems is a syntax issue or a need to do something different with the library when using arrays:

what is the proper syntax for passing a reference to a nested array handle to be used by another function? (I'm trying to recursively parse directories of an SD card and send the results back)

Second, once I pass the array, I'm trying to populate it with nested objects describing the files/directories under that array. Thus I'm not sure if it's my reference syntax that is wrong or the way I'm trying to create the object that is wrong. I'm getting: 'no matching function call to Arduino::JsonArray::createNestedObject(String&)'

Third is just a curiosity as the code hasn't made it past those errors to see if this works, will it or do you have to pass a character buffer?
  String jsonOut;
  root.printTo(jsonOut);

bblanchon, 2016-08-26:
Hi Scott.

To pass a reference to a `JsonArray`, just add a `&` in the function signature:

```c++
void myFunction(JsonArray& array) {
  //
}
``` 

There is one caveat: you must be sure that the array is in memory (ie the `JsonBuffer` must be in memory) at the time this function is called.

`JsonArray::createNestedObject()` doesn't take any argument.

A `String` would work but a preallocated `char[]` is better when you have very few memory:

```c++
char jsonOut[256];
root.printTo(jsonOut, sizeof(jsonOut));
```

treii28, 2016-08-26:
Ahh, so chances are my problem is scope related and I need to make the Buffer global?

treii28, 2016-08-26:
Still getting the same error:

`
DynamicJsonBuffer jsonBuffer;

void listDirJSON(JsonArray& jArr, String path, boolean descend) {

  File dir = SD.open((char *)path.c_str());
  dir.rewindDirectory();

  for (int cnt = 0; true; ++cnt) {
    File entry = dir.openNextFile();
    if (!entry)
      break;

    String filename = entry.name();
    // skip dot files
    if (filename.startsWith("."))
      break;

    JsonObject& item = jArr.createNestedObject(filename);
    if (entry.isDirectory()) {
      item["type"] = "dir";
      if (descend == true) {
        JsonArray& content = item.createNestedArray("content");

        String fullpath = path + filename + "/";
        DBG_OUTPUT_PORT.println("descending path: " + fullpath);
        listDirJSON(content, fullpath, false);
      }
    } else {
      item["type"] = "file";
      item["size"] = String(entry.size(), DEC);
    }

    entry.close();
  }
  dir.close();
}

void printFS() {
  JsonObject& root = jsonBuffer.createObject();

  root["name"] = "/";
  root["type"] = "dir";
  JsonArray& content = root.createNestedArray("content");
  listDirJSON(content, "/", false);

  String jsonOut;
  root.printTo(jsonOut);
  returnJSON(jsonOut);
}
`

treii28, 2016-08-26:
nevermind, just spotted it. Didn't quite grasp your comment when you said 'it doesn't take an argument' - duh, so it's looking for a definition where it takes a string and isn't finding one. *smacks forehead*

closed 2016-08-26 by treii28