bblanchon/ArduinoJson #341: Support for saving location

devinrayolsen, 2016-08-28:
First let me say how much I love this lib!! I'm using it to both build an menu system for my LCD that also containers system variables that all gets saved and loaded from SD card (Ardunio Mega R3).

That said, when working with complex nested objects (10 wide / 4 levels deep) it would be greatly helpful to be able to have a way to quickly reference back to a location within a complex object. For instance, in my use case I touch(read) an object then list it's contents to my LCD. Using a global int (currentMenuIndex) and a scroll function, I can scroll through an objects subobjects. However, when it comes time to then move into one of these sub objects and read it's contents, it requires me to know the full path back to this newly touched object, so on and so forth.

I've tried keeping a global JsonObject (I know I know local local local) as a storing location (and as a pointer) for where I currently am at within my complex object, but as you are aware, you can't re-assign one JsonObject to another like that.

I've also tried to keep an array of strings that contains the name of each object I've touched (like a history) so that I could use that array's items as a way to get back to my current location a complex object.. Alas I know of no way to dynamically use X number of brackets[] that contain X number of history names (reflection if you will) to get me back to my current location.

Is there already a way I'm just not aware of.. and if not, might I purpose something like the following.

Say we have a 4 level deep nesting of objects:
```
JsonObject& root = jsonBuffer.createObject();
	root["level one"];
	JsonObject& lvlOne = root.createObject("level one"); 

	    lvlOne["level two"];
	    JsonObject& lvlTwo = lvlOne.createObject("level two");

	        lvlTwo["level three"];
	        JsonObject& lvlThree = lvlTwo.createObject("level three");

	            lvlThree["level four"];
	            JsonObject& lvlFour = lvlThree.createObject("level four");
```
Instead of getting to lvlFour by:
`root["level one"]["level two"]["level three"]["level four"]`

it would be simply amazing to be able query our back to a location via again a self maintained history string array by that we concat and then pass like so:

`root.queryObject("level one/level two/level three/level four");`
and even
`root.queryArray("level one/level two/level three/level four");`

Possible?
Thanks!

bblanchon, 2016-08-28:
> as you are aware, you can't re-assign one JsonObject to another like that.

I'm not sure what you meant by that, but if you're talking about a `JsonObject&`, yes it cannot be reassigned, as any C++ reference. But you can switch to a pointer:

```c++
JsonObject current = &root["a"]["b"].as<JsonObject>();
```

That means that you can quickly implement the feature you ask for with an `std::map<std::string,JsonObject*>`.
However, if I were to do that, I would index with an integer instead of a string.
That would mean you could just use an array of `JsonObject*`.

That being said, let's take a little distance and let's ask why we need this.

As [it's said in the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-reuse-a-jsonbuffer), if you think you need a global JsonBuffer, it's because you're not using the library the way it's meant to.
Indeed ArduinoJson is just a **serialization** library; it's not intended to be a framework that you use as a backbone for your app.
Getting back to your issue, you're asking something that is not related to serialization, and this is a sign that you're using ArduinoJson for something else.

My advice is to implement your own class for the menu, and only use ArduinoJson to hydrate them from the JSON file.
If you follow that, you'll probably get back to the [cannonical example](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library) and instantly stop having a headache because of ArduinoJson.

devinrayolsen, 2016-08-28:
Ah, yes thank you for the snip, I missed that! That's a huge help!

As for the rest, 110% in agreement I'm using this in an unorthodox approach. Its just the draw to being able to wrap all my app's settings and objects into a single string, save it on my MicroSD and load it back up with ease in say the case of a power outage or even saving unique versions of said apps settings. You've built something really awesome here!

Thanks for the quick reply; my request above seems silly now that I know the above snip!

bblanchon, 2016-08-29:
You're welcome @devinrayolsen, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-08-29 by bblanchon