bblanchon/ArduinoJson #343: Recursively entering Json object based on string

tttapa, 2016-08-28:
Hello, I've got the following json object:
```JSON
{
    "ESP_IN_PC_OUT":{
        "track":{
            "1":{
                "mute":{
                    "type":"LED",
                    "pins":[4]
                }
            },
            "2":{
                "mute":{
                    "type":"LED",
                    "pins":[5]
                }
            }
        }
    }
}
```
I want to enter the object, depending on a string of the format "/index1/index2/index3 ...". For example, "/track/2/mute" should return a Json object like this:
```JSON
{
    "type":"LED",
    "pins":[5]
}
```
I'm using an ESP8266. This is the code I've tried:
```Arduino
#include <ArduinoJson.h>

char *jsonstr = "{\n    \"ESP_IN_PC_OUT\":{\n        \"track\":{\n            \"1\":{\n                \"mute\":{\n                    \"type\":\"LED\",\n                    \"pins\":[4]\n                }\n            },\n            \"2\":{\n                \"mute\":{\n                    \"type\":\"LED\",\n                    \"pins\":[5]\n                }\n            }\n        }\n    },\n\n\n    \"ESP_OUT_PC_IN\":{\n        \"analog\":{\n        },\n        \"digital\":{\n            \"16\":\"/track/1/mute\"\n        }\n    }\n}"; 

void setup() {
  Serial.begin(115200);

  DynamicJsonBuffer jsonBuffer;

  JsonObject& root = jsonBuffer.parseObject(jsonstr);
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  Serial.println((const char*) root["ESP_IN_PC_OUT"]["track"]["2"]["mute"]["type"]); // prints "LED"
  Serial.println();
  
  char *address = "/track/2/mute";
  char *tmp = strtok(address,"/");
  JsonObject& tmpJson = root["ESP_IN_PC_OUT"];
  bool success = true;
  while(tmp != NULL && success == true){
    Serial.println(tmp);
    if(tmpJson.containsKey(tmp)){
      tmpJson = tmpJson[tmp];        // Error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
      tmp = strtok(NULL,"/");
    } else {
      success = false;
    }
  }
  if(success == true){
    Serial.println((const char*)tmpJson["type"]);
  } else {
    Serial.println("Address not found");
  }
}

void loop() {
}
```
However, this gives me the following error message:
```
JSON_OSC_settings2:26: error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'

       tmpJson = tmpJson[tmp];

               ^

In file included from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson.hpp:12:0,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson.h:8,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/ArduinoJson.h:8,

                 from C:\Users\Pieter\Documents\Arduino\JSON_OSC_settings2\JSON_OSC_settings2.ino:1:

C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson/JsonObject.hpp:38:7: note: 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)' is implicitly deleted because the default definition would be ill-formed:

 class JsonObject : public Internals::JsonPrintable<JsonObject>,

       ^

In file included from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson/JsonArray.hpp:13:0,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson.hpp:11,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson.h:8,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/ArduinoJson.h:8,

                 from C:\Users\Pieter\Documents\Arduino\JSON_OSC_settings2\JSON_OSC_settings2.ino:1:

C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson/Internals/ReferenceType.hpp:32:18: error: 'ArduinoJson::Internals::ReferenceType& ArduinoJson::Internals::ReferenceType::operator=(const ArduinoJson::Internals::ReferenceType&)' is private

   ReferenceType& operator=(const ReferenceType&);

                  ^

In file included from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson.hpp:12:0,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson.h:8,

                 from C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/ArduinoJson.h:8,

                 from C:\Users\Pieter\Documents\Arduino\JSON_OSC_settings2\JSON_OSC_settings2.ino:1:

C:\Users\Pieter\Documents\Arduino\libraries\ArduinoJson-master/include/ArduinoJson/JsonObject.hpp:38:7: error: within this context

 class JsonObject : public Internals::JsonPrintable<JsonObject>,

       ^

exit status 1
use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
```
How can I solve this?

Thanks in advance!
Pieter

bblanchon, 2016-08-29:
Hello Pieter,

Indeed, you cannot reassign a `JsonObject&`.

One solution is to use a pointer instead.

```c++
JsonObject* tmpJson = &root["ESP_IN_PC_OUT"].as<JsonObject>();
```

You can also use a `JsonVariant` which will act as a wrapper around the pointer.

```c++
JsonVariant tmpJson = root["ESP_IN_PC_OUT"];
```

Regards,
Benoit

bblanchon, 2016-08-29:
PS: I added this to the FAQ: [How to assign a JsonArray or JsonObject?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-assign-a-jsonarray-or-jsonobject)

tttapa, 2016-08-29:
It works perfectly now, thank you.
And thanks a lot for this great library!

closed 2016-08-29 by tttapa