bblanchon/ArduinoJson #347: Global Dynamic Buffer getting corrupted

themindfactory, 2016-09-08:
So I have read about avoiding pitfalls and am totally on-board with the concept of scope and the buffers.... however I have created one dynamic buffer that I wish to always be able to access no matter what function if being called, this contains "system setup" info and needs to always be around... but I do have issues with this buffer and the JSON object within it getting corrupted.....

Any cause for alarm in what I have done??

bblanchon, 2016-09-09:
Using a global `JsonBuffer` is always a bad idea; you're really asking for troubles.

If you're sure you want to go this way, I'll need some clarification on what you mean by "corrupted" and see some code.

See:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-shouldnt-i-use-a-global-jsonbuffer)
* [FAQ: How to reuse a JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-reuse-a-jsonbuffer)

themindfactory, 2016-09-09:
this could be a good lesson for all then!

save was defined as globally as:
```
DynamicJsonBuffer jsonSaveBuffer;
JsonObject& save = jsonSaveBuffer.createObject();
```

saveData() loops through a structure looking for valid commands and what to do with them...
when saveData() was entered I had a debug print statement doing a print and it would look like:
{"garbage":"garbage","garbage":"garbage","command 1","command 1"}
as SaveData is repeatedly called commands are added and the garbage stays the same,
each time the device is restarted the garbage changes to random values and amount, and the new save data is appeneded to the end...
```
void saveData(String cmd, String value) {
 // debug went here
  for (int i = 0; i < numCommands; ++i) {
    if (cmd == commandLookup[i].command) {
      if (commandLookup[i].save) {

        // we save the data here
        save[cmd] = value;

       // saves JSON string to file
       File f = SPIFFS.open("/data.txt", "w");
        String s = "";
        save.prettyPrintTo(s);
        f.println(s);
        f.close();
      }
}

```

loadData() parses the save JSON string and enters it back into save object, this is only done once at startup
```
void loadData() {
  DynamicJsonBuffer jsonSaveBuffer;

  File f = SPIFFS.open("/data.txt", "r");
  String s = f.readString();
  JsonObject& n = jsonSaveBuffer.parseObject(s);
  for (JsonObject::iterator it = n.begin(); it != n.end(); ++it) {
      save[it->key] = it->value;
  }
  f.close();
}
```

bblanchon, 2016-09-09:
It seems that your `DynamicJsonBuffer` keeps growing and growing until memory is exhausted.

I told you, a global `JsonBuffer` is the recipe for failure.

Why don't you use something else to store the strings (like a std::map) and use ArduinoJson for serialization only?

See:

* [FAQ: What's the best way to use the library?](#whats-the-best-way-to-use-the-library)

themindfactory, 2016-09-09:
Well it does grow but only to the tune of 20 entries max all with about a payload of maybe 32 bytes so its not large. I need to look at the std::map thing as I have not done that yet, and just new to C++, even though I am very confident with C, 20 years of C :-) Wonder if I can find some good real simple snippets to store my data.... hmmm

Thanks!

themindfactory, 2016-09-09:
I think this is enough to get me going... http://www.cplusplus.com/reference/map/map/map/

closed 2016-09-10 by themindfactory