bblanchon/ArduinoJson #348: how can i send and receive json via WebSocketsServer.h?

piux, 2016-09-09:
im using WebSocketsServer.h from Links2004 with an esp8266 (model esp-07) arduino ide, i just stumbled with this library and i wanted to test if i could send a simple json with this.

in order to send a txt you use this function.
`
void WebSocketsServer::sendBIN(uint8_t num, uint8_t * payload, size_t length, bool headerToPayload) {
    if(num >= WEBSOCKETS_SERVER_CLIENT_MAX) {
        return;
    }
    WSclient_t * client = &_clients[num];
    if(clientIsConnected(client)) {
        sendFrame(client, WSop_binary, payload, length, false, true, headerToPayload);
    }
}`

num is the number of the client connected to the socket and the uint8_t * payload argument is where you send the message.

a simple example would be ' webSocket.sendTXT(num, "connected");'

im stuck at the function root.printTo(Serial);.

i dont understand how does this work or send the data to the serial usb, also im not using the serial port, im using the websocket.

im kinda lost on were to start or how this prints.

thank you for your time.

JeroenRuiten, 2016-09-09:
I've more or less the same setup. You need to write it to a buffer and then give that buffer to the webSocket.sendTXT.

See more in the [documentation](https://github.com/bblanchon/ArduinoJson/wiki/Encoding%20JSON#step-3-generate-the-json-string)

Like this

`char buffer[255];` Be sure your buffer is large enough to hold the string
`root.printTo(buffer, sizeof(buffer);` write the json object to the string
`webSocket.sendTXT(num, buffer);` pass the string to websocket's function

Just be aware you can't declare any variables in a switch statement, if there are no brackets in the case. Better to point to a function and do it all there. (I say this to avoid confusion. You are likely to react to a incoming message. This is inside a switch statement)

I know it is't the most efficient way, but as far as I know it's the only way.

bblanchon, 2016-09-09:
Another alternative would be to write an adapter on top of `WebSocketServer` that implements the `Print` interface.

The Bag of Tricks contains a [Buffered Output](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks#buffered-output) that would be a good starting point.

piux, 2016-09-09:
thanks it worked!

closed 2016-09-09 by piux