bblanchon/ArduinoJson #349: Conformance report for reference

miloyip, 2016-09-09:
https://github.com/miloyip/nativejson-benchmark/blob/master/sample/conformance_ArduinoJson%20(C%2B%2B).md

bblanchon, 2016-09-09:
Hi @miloyip,

Thank you very much for this report.
Unfortunately, most of the problems are imposed by designed decisions:

* parser aims to be very tolerant (quotes are optional most of the time and trailing chars are ignored)
* escape sequence `\u` and `\x` are not supported (would increase code size for nothing)
* nesting is limited by default to 10 (security feature to prevent a avoid stack overflow)

However, I'll dig into each problem to see if it's a legitimate non-compliancy or not.

Again, thank you very much, this is gold.

bblanchon, 2016-09-25:
I fixed the bug in float parsing and increased the default nesting limit.
Thanks again.

closed 2016-09-25 by bblanchon