bblanchon/ArduinoJson #351: Coding Question

tigergroup, 2016-09-09:
I have been working on this for over a day and seem to not get this working correctly.  I have a json that I parsed.  I need to take the current password in the json, encrypted it and put it back in the json.

json:

{
"wifi": {
    "ssid": "somessid",
    "password": "somepassword"
  },
}

convert to:

{
"wifi": {
    "ssid": "somessid",
    "password": "AGjsdfDv4Sf3f"
  },
}

I tried something like this:

StaticJsonBuffer<MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE> jsonBuffer;
  JsonObject& parsedJson = jsonBuffer.parseObject(buf);
  if (!parsedJson.success()) {
    this->_interface->logger->logln(F("✖ Invalid JSON in the config string"));
    return;
  }

    JsonObject& jsonWifi = parsedJson["wifi"];
    int strLength = strlen(jsonWifi["password"]);
    char unencryptedPassword[strLength]; 
    strcpy(unencryptedPassword, jsonWifi["password"]);

    char encryptedWifiPasswordEncoded[MAX_WIFI_PASSWORD_LENGTH];
    _encryptPassword(encryptedWifiPasswordEncoded, (byte*)unencryptedPassword, strlen(unencryptedPassword));
    jsonWifi.set("password",encryptedWifiPasswordEncoded);

Can anyone guide me on what I am doing wrong?

scott

bblanchon, 2016-09-09:
This should work:

```c++
const char* clearPassword = parsedJson["wifi"]["password"];

char securePassword[MAX_WIFI_PASSWORD_LENGTH];
_encryptPassword(securePassword, (byte*)clearPassword, strlen(clearPassword));

parsedJson["wifi"]["password"] = securePassword;
```

However, I'm a little concerned about the extra comma at the end of the "wifi" object in the input.
What does `parsedJson.success()` says?

tigergroup, 2016-09-09:
got this compiler error for esp8266 chip:

error: passing 'const ArduinoJson::JsonObjectSubscript<const char*>' as 'this' argument of 'typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::JsonObject::CanSet<T>::value, ArduinoJson::JsonObjectSubscript<TKey> >::type& ArduinoJson::JsonObjectSubscript<TKey>::operator=(T) [with T = char*; TKey = const char*; typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::JsonObject::CanSet<T>::value, ArduinoJson::JsonObjectSubscript<TKey> >::type = ArduinoJson::JsonObjectSubscript<const char*>]' discards qualifiers [-fpermissive]

 parsedJson["wifi"]["password"] = securePassword;

bblanchon, 2016-09-09:
Hmmm... that's strange; I'll open an issue on that topic.

Anyway, here is the workaround:

```c++
JsonObject& wifi = parsedJson["wifi"];
const char* clearPassword = wifi["password"];

char securePassword[MAX_WIFI_PASSWORD_LENGTH];
_encryptPassword(securePassword, (byte*)clearPassword, strlen(clearPassword));

wifi["password"] = securePassword;
```

bblanchon, 2016-09-09:
Interesting, did you have an error from the compiler?

bblanchon, 2016-09-09:
I just realized that you're using a `StaticJsonBuffer` that is probably too big to fit in the stack.

You should switch to a `DynamicJsonBuffer`.

closed 2016-09-16 by bblanchon