bblanchon/ArduinoJson #353: How to parse array of objects?

abhisheksafui, 2016-09-15:
I searched through all examples but could not find how to parse array of objects like : 
{
   "message_type" : 1, 
    "requests" : [
      { "type" : 12 , "value" : "abc"},
      { "type" : 15, "value" : "xyz"}, 
    ]
}

Can you please explain if it is possible to parse this. I will be populaiting C list nodes based on "requests".

bblanchon, 2016-09-16:
```c++
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(intput);
JsonArray& requests = root["requests"];
for (auto& request : requests) {
   int type = request["type"];
   const char* value = request["value"];
}
```

I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

abhisheksafui, 2016-09-16:
Thanks for the reply and the library. I will try it out.  But neither the decoding section nor the Examples section documented this example. You could include this example in it. Otherwise documentation is fine.

abhisheksafui, 2016-09-16:
Thanks a lot it worked, it was just a C programmers difficulty in understanding iterator, which yu have  well documented.  Another doubt: Placing StaticJsonBuffer<500> replyBuffer;  inside switch case is giving following error. 
interface:47: error: crosses initialization of 'ArduinoJson::StaticJsonBuffer<500u> replyBuffer'
      StaticJsonBuffer<500> replyBuffer;
                            ^
exit status 1
jump to case label [-fpermissive]

Placing inside if, that is inside the case ABC:  again works well.  
Never faced error like this in C. Might be some C++ related. Time to learn.

bblanchon, 2016-09-19:
You should define a scope in the `case`, or better extract a function.

closed 2016-09-19 by bblanchon