bblanchon/ArduinoJson #354: Prettyfier uses undefined behaviour to sequence writes

Dwergi, 2016-09-16:
The entirety of Prettyfier uses this pattern:

     size_t writeBlockClose(uint8_t c) {    
          return unindentIfNeeded() + _sink.write( c );    
     }

This makes an assumption that unindentIfNeeded is called first, but this assumption is wrong, because `+` is not a sequence point. C++ makes no guarantees about evaluation order in this case, and in VS2015, `_sink.write` is called first.

To guarantee order of evaluation, all of these functions should be split across two lines:

      size_t writeBlockClose(uint8_t c) {
	    size_t written = unindentIfNeeded();
	    return written + _sink.write( c );
      }

Dwergi, 2016-09-16:
This also applies to `Print`, which does this:

`size_t println() { return write('\r') + write('\n'); }`

This also causes the indents to occur before the `\r`, because of the newline checking behaviour in `IndentedPrint`.

bblanchon, 2016-09-16:
You're right, I'll fix that ASAP.

closed 2016-09-17 by bblanchon