bblanchon/ArduinoJson #355: parseObject() failed.  However online validating is ok!

MHz000, 2016-09-18:
When I parse the following JSON I got "parseObject() failed". Validating the same object with different online server gave no error. 
{
"error":false,
"apiversion":"1.3.0",
"year":2016,
"month":9,
"day":18,
"dayofweek":"Sunday",
"datechanged":false,
"lon":9.160000,
"lat":47.730000,
"tz":1,

"sundata":[
            {"phen":"BC", "time":"05:34"},
            {"phen":"R", "time":"06:05"},
            {"phen":"U", "time":"12:17"},
            {"phen":"S", "time":"18:29"},
            {"phen":"EC", "time":"19:00"}],

"moondata":[
            {"phen":"U", "time":"01:23"},
            {"phen":"S", "time":"07:53"},
            {"phen":"R", "time":"19:38"}],

"prevmoondata":[
            {"phen":"R","time":"19:05"}],

"closestphase":{"phase":"Full Moon","date":"September 16, 2016","time":"20:05"},
"fracillum":"96%",
"curphase":"Waning Gibbous"
}
parseObject() failed

On the opposite parsing the JSON object contained in the JsonParserExample "Copyright Benoit Blanchon 2014-2016" reading:
{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}
results always in an error message  :-((
Any hint, any help?

bblanchon, 2016-09-19:
What platform is it?
Did you have a look at [compatibility issues](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility%20issues)?

MHz000, 2016-09-19:
due to receiving so sparse information from the parser, just pasing failed,
I run into a memory problem. Now after switching to an ESP8266 and allocate
much more memory the problen ist solved.
However one issue remains. The JSON object from the example is parsed with
no problem where as an online check by different JSON validators mark it as
faulty.

Anyway thanks for supplying the JSON lib

closed 2016-09-19 by MHz000

bblanchon, 2016-09-19:
Indeed, the only information you get is `true` or `false`, you cannot tell if the input is bad of if the allocation failed.
Sorry about that, but this is to keep the code as small as possible.
ArduinoJson is not meant to be a JSON validator anyway.

BTW, I just tested the example with http://jsonlint.com/, it says "Valid JSON", what did you try exactly?

bblanchon, 2016-09-19:
See:

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

MHz000, 2016-09-19:
Stupid me! I've used the line from the source code for testing :-(
Sorry for all the fuss

bblanchon, 2016-09-19:
No big deal.
Thanks for using ArduinoJson.