bblanchon/ArduinoJson #356: Undefined reference to __cxa_guard_acquire and __cxa_guard_release

jonblack, 2016-09-20:
When I compile the `JsonParserExample` I get the following error:

```
build-mega-atmega2560/example.ino.o: In function `ArduinoJson::JsonArray::invalid()':
example.ino:(.text._ZN11ArduinoJson9JsonArray7invalidEv[ArduinoJson::JsonArray::invalid()]+0xc): undefined reference to `__cxa_guard_acquire'
example.ino:(.text._ZN11ArduinoJson9JsonArray7invalidEv[ArduinoJson::JsonArray::invalid()]+0x28): undefined reference to `__cxa_guard_release'
build-mega-atmega2560/example.ino.o: In function `ArduinoJson::JsonObject::invalid()':
example.ino:(.text._ZN11ArduinoJson10JsonObject7invalidEv[ArduinoJson::JsonObject::invalid()]+0xc): undefined reference to `__cxa_guard_acquire'
example.ino:(.text._ZN11ArduinoJson10JsonObject7invalidEv[ArduinoJson::JsonObject::invalid()]+0x28): undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status
/home/jon/.arduino_mk/Arduino.mk:1416: recipe for target 'build-mega-atmega2560/arduino-steem.elf' failed
make: *** [build-mega-atmega2560/arduino-steem.elf] Error 1
```

I remember a similar problem with the Ethernet library a while back and found the issue for it (https://github.com/esp8266/Arduino/issues/574). Perhaps that can help.

bblanchon, 2016-09-20:
I would be very interested to know if you find a proper workaround so that I could add it to [Compatibility Issues](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility%20issues).
It's probably a flag missing in the invocation of the linker.

jonblack, 2016-09-23:
I figured out the problem. I had the following in my Makefile:

    CXXFLAGS_STD += -std=gnu++11

Removing this fixed the problem.

bblanchon, 2016-09-23:
I don't like this because it mean you won't be able to use C++11 features in your code.

Can you try to add this flag instead: 

> `-fno-threadsafe-statics`
> Do not emit the extra code to use the routines specified in the C++ ABI for thread-safe initialization of local statics. You can use this option to reduce code size slightly in code that doesn't need to be thread-safe.

jonblack, 2016-09-24:
Apparently Arduino defaults to `c++14` and I was appending `gnuc++11`, so it got totally messed up. Should be fine without it.

closed 2016-09-25 by bblanchon

satishIoT, 2017-03-13:
CXXFLAGS_STD += -std=gnu++11  ; Could you pl give steps where precisely this correction to be made. Im finding very hard to make out.

bblanchon, 2017-03-13:
Hello Satish,

If you use the Arduino IDE, you need to edit the `platform.txt` of the board you're using:

* The file is located at `%LOCALAPPDATA%\Arduino15\packages\<brand>\hardware\<board>\<version>\platform.txt`
* The line to change is `compiler.cpp.flags`

I edited the [FAQ](https://bblanchon.github.io/ArduinoJson/faq/compilation-fails-device-crashes-nothing-on-serial-console/#error-undefined-reference-to-cxaguardacquire-and-cxaguardrelease), so that it'll be easier to find.

Please confirm this solves your problem.

Regards,
Benoit

satishIoT, 2017-03-13:
Thanks Benoit , for the speedy reply. Will try out and get back to you at
once.

regards

satish

satishIoT, 2017-03-13:
Hi Benoit

Located the path for platform.txt for esp8266 module found the following
lines .
compiler.cpp.flags=-c -Os -g -mlongcalls -mtext-section-literals
-fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD
-ffunction-sections -fdata-sections

Found the line as
compiler.cpp.flags=-c -Os -g -mlongcalls -mtext-section-literals
-fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD
-ffunction-sections -fdata-sections

Removed the line:  -std=c++11
compiler.cpp.flags=-c -Os -g -mlongcalls -mtext-section-literals
-fno-exceptions -fno-rtti -falign-functions=4 -MMD -ffunction-sections
-fdata-sections

Saved the file and opend Arduino 1.6.4 version IDE

From examples ArduinoJson took the file JsonParserExample.ino file and
recompiled ... This gave me the following error

Arduino: 1.6.4 (Windows 8.1), Board: "Generic ESP8266 Module, Serial, 80
MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

In file included from
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:247:0,
                 from JsonParserExample.ino:10:
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/HardwareSerial.h:85:27:
warning: override controls (override/final) only available with -std=c++11
or -std=gnu++11 [enabled by default]
         int available(void) override;
                           ^
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/HardwareSerial.h:86:22:
warning: override controls (override/final) only available with -std=c++11
or -std=gnu++11 [enabled by default]
         int peek(void) override;
                      ^
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/HardwareSerial.h:87:22:
warning: override controls (override/final) only available with -std=c++11
or -std=gnu++11 [enabled by default]
         int read(void) override;
                      ^
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/HardwareSerial.h:89:24:
warning: override controls (override/final) only available with -std=c++11
or -std=gnu++11 [enabled by default]
         void flush(void) override;
                        ^
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/HardwareSerial.h:90:29:
warning: override controls (override/final) only available with -std=c++11
or -std=gnu++11 [enabled by default]
         size_t write(uint8_t) override;
                             ^
In file included from JsonParserExample.ino:10:0:
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:275:48:
error: 'nullptr' was not declared in this scope
     const char* server1, const char* server2 = nullptr, const char*
server3 = nullptr);
                                                ^
C:\Users\Satish\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:275:79:
error: 'nullptr' was not declared in this scope
     const char* server1, const char* server2 = nullptr, const char*
server3 = nullptr);

   ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Have spent more then a day to sort the problum still getting same error .
Appreciate your help to solve the issue.

regards

satish

bblanchon, 2017-03-13:
Satish, you should add `-fno-threadsafe-statics` and keep `-std=c++11`.

Explanation:

1. `-std=c++11` enables support for the new language features introduced in the C++11 standard, like `nullptr`
2. `-fno-threadsafe-statics` disables the mutex around initialization of local static variables, which is useless when you program is single threaded.

satishIoT, 2017-03-13:
Aaha finally done compiling :) . Thank you very much had understood totally
wrong .

jonblack, 2017-03-14:
Just to clarify the cause (for me). I was **appending** `-std=gnu++11` to `CXXFLAGS_STD`. Because it defaults to `-std=gnu++14`, I had two standards defined which of course isn't gonna work.