bblanchon/ArduinoJson #358: MessagePack Support?

mahmed8003, 2016-09-24:
Is it possible to have MessagePack support.
Instead of printing JSON string to buffer using this method
`size_t prettyPrintTo(char *buffer, size_t bufferSize)`
can we have another method like
`size_t msgPackPrintTo(char *buffer, size_t bufferSize)`

Use case:
- Final buffer size could be smaller and it can reduce the network data usage where sensor sending lot of data in json string.
- Instead of storing plain JSON string to SPIFFS we can store small msgPack buffer. It will reduce file size.

bblanchon, 2016-09-25:
This sound like a good idea, and that's 100% aligned with the philosophy of ArduinoJson.
However, I'm afraid most users would prefer the official implementation instead of ArduinoJson...

Also, it looks like there will be a lot of testing, so I'd like to see how much traction this feature gets before writing a lot of code.

Let's see how many upvotes 👍 we get on this feature request.

heman4t, 2016-09-26:
When its about...
> Final buffer size could be smaller and it can reduce the network data usage where sensor sending lot of data in json string.

I upvote. For me size is equally important to performance in micro controllers.

sticilface, 2016-09-26:
I'd be keen on this.  reducing the buffer size is great from my end!  limited ram and all..

cosmikwolf, 2017-02-04:
upvote. I want to use arduino JSON to store data on local storage, but it is not efficiently packed. Would love a way to reduce the output size. Even just storing variables in their hex format would be good.

cosmikwolf, 2017-02-04:
ArduinoJSON is miles ahead of the msgPack arduino implementation, would way rather use ArduinoJSON for this purpose.

bblanchon, 2017-02-11:
Branch 5.8.x seems to be quite stable now.
I'll now focus on this feature.

Matheus-Garbelini, 2018-02-19:
@bblanchon there's a new PR of mine that includes an initial support to messagepack.
https://github.com/bblanchon/ArduinoJson/pull/682

bblanchon, 2018-06-07:
MessagePack is finally available in [ArduinoJson 6.0.0](https://arduinojson.org/news/2018/06/07/version-6-0-0/?utm_source=github&utm_medium=issues)

commit fc2e3a4a (2018-05-29): Added `serializeMsgPack()` and `measureMsgPack()` (closes #358)
closed 2018-06-07 by bblanchon