bblanchon/ArduinoJson #359: Exception (9) after second invoke of parseObject()

MHz000, 2016-09-25:
My environment: Arduino 1.6.9 with generic ESP8266 moule using external power

first parsing always succeeded however next call of parseObject() crashes with Exeption (9) see end of post.
Time between first and second call ~20 seconds as coded by:

    delayMillis= millis()+20000; 
    do {
     yield();
     } while (millis() < delayMillis);                 // ==> no WDT interruption

**What's wrong with my code?** 

...
StaticJsonBuffer<1000> jsonBuffer;                    // Memory pool for JSON object tree.
char json[900];                                       // JSON object char array, filld with server response 
...
JsonObject& root = jsonBuffer.parseObject(json); // parse JSON object 
    if (!root.success()) 
      {Serial.println("parseObject() failed");}      // test if parsing succeeded.
      else  {Serial.println("parseObject() success");
             digitalWrite(LED_PIN, HIGH);}           // turn off LED 
            
    Serial.println(); Serial.println("extracted from JSON object"); 
... 
--------------------------------------------------------------------------------------
Header and JSON oject retrieved from server looks like:

HTTP/1.1 200 OK
Date: Sun, 25 Sep 2016 08:53:24 GMT
Server: Apache
Access-Control-Allow-Origin: *
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json

292
{
"error":false,
"apiversion":"1.3.0",
"year":2016,
"month":9,
"day":25,
"dayofweek":"Sunday",
"datechanged":false,
"lon":9.160000,
"lat":47.730000,
"tz":1,

"sundata":[
            {"phen":"BC", "time":"05:44"},
            {"phen":"R", "time":"06:14"},
            {"phen":"U", "time":"12:15"},
            {"phen":"S", "time":"18:14"},
            {"phen":"EC", "time":"18:45"}],

"moondata":[
            {"phen":"R", "time":"00:18"},
            {"phen":"U", "time":"07:55"},
            {"phen":"S", "time":"15:27"}],

"closestphase":{"phase":"Last Quarter","date":"September 23, 2016","time":"10:56"},
"fracillum":"28%",
"curphase":"Waning Crescent"
}

---------------------------------------------------------------------------------------
first successfull parsing and error report after second call:

***enter main loop***
WiFi Connected
***client connection ok***
server connected
GET request ok!
extracted from server header
day: 25
mon : 9
yea: 2016
hou: 5
min: 46
Systemzeit h:m 5:46 = 346
parseObject() success

extracted from JSON object
25.9.2016
05:44 = 344
06:14 = 374
12:15 = 735
18:14 = 1094
18:45 = 1125

***enter main loop***
***client connection ok***
server connected
GET request ok!
extracted from server header
day: 25
mon : 9
yea: 2016
hou: 5
min: 47
Systemzeit h:m 5:47 = 347

Exception (9):
epc1=0x40204a19 epc2=0x00000000 epc3=0x00000000 excvaddr=0x30223a22 depc=0x00000000

ctx: cont 
sp: 3ffefcc0 end: 3ffeff50 offset: 01a0

>>>stack>>>
3ffefe60:  3ffe85ec 00000003 3ffeeefc 40204129  
3ffefe70:  3ffe88ac 0000015b 3ffeeefc 4020238a  
3ffefe80:  3ffe85ec 3ffeee14 3ffeeefc 40204154  
3ffefe90:  3ffe85ec 3ffee64c 3ffeeefc 0000015b  
3ffefea0:  3ffe85ec 3ffee64c 3ffeeefc 40202cfd  
3ffefeb0:  3ffee9d0 3ffee64c 3ffee64c 4010040a  
3ffefec0:  00000000 00000000 00000000 00000000  
3ffefed0:  00000000 00000000 00000000 00000000  
3ffefee0:  00000000 00000000 00000000 00000000  
3ffefef0:  00000000 00000000 00000000 00000000  
3ffeff00:  00000000 00000000 3ffeeefc 40204154  
3ffeff10:  00000012 feefeffe 3ffeeefc 40204178  
3ffeff20:  00000000 00000000 00000001 3ffeef28  
3ffeff30:  3fffdad0 00000000 3ffeef20 402047fc  
3ffeff40:  feefeffe feefeffe 3ffeef30 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
ø

bblanchon, 2016-09-25:
```c++
StaticJsonBuffer<1000> jsonBuffer;
char json[900];
```

Those two are probably too big to fit in the stack.
AFAIK, the stack is limited to 4KB on ESP8266, but the rest of it is likely to be taken by the rest of the program.

See:

* [FAQ: Why does my device crash or reboot?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-does-my-device-crash-or-reboot)

MHz000, 2016-09-25:
May be. However the very first invocation returns JSON values without any
problem. Is there a way to figure out what buffer size for a given
JSON-object will bee needet. The server respons has always the same
structur and length.

Thanks for quick respons.

bblanchon, 2016-09-25:
See:

* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)
* [FAQ: What are the common sizes for JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#what-are-the-common-sizes-for-jsonbuffer)
* [JsonBuffer size calculator](https://rawgit.com/bblanchon/ArduinoJson/master/scripts/buffer-size-calculator.html)

MHz000, 2016-09-25:
Thank you for guiding me to the JSON Buffer size calculator
Checking my JSON string with "JSON Buffer size calculator" = 736.  That's
3/4 from allocated buffer in StaticJsonBuffer<1000> jsonBuffer.
Compiler reports too  plenty of memory aviable:
"Der Sketch verwendet 236.613 Bytes (*54%*) des Programmspeicherplatzes.
Das Maximum sind 434.160 Bytes.
Globale Variablen verwenden 34.604 Bytes (*42%*) des dynamischen Speichers,
47.316 Bytes für lokale Variablen verbleiben. Das Maximum sind 81.920
Bytes."
The good news: results of first call of JsonObject& root =
jsonBuffer.parseObject(json); are complet ok
The bad news:  the following calls yields to:
Exception (9):
epc1=0x402049dd epc2=0x00000000 epc3=0x00000000 excvaddr=0x30223a22
depc=0x00000000
ctx: cont
sp: 3ffefc80 end: 3ffeff10 offset: 01a0

I'm sure something is incorrect in my codeing. Btw: the server HTTP header
from the second call is correct received and decoded. I'm confused :-(

bblanchon, 2016-09-26:
[The Arduino IDE shows](https://github.com/esp8266/Arduino/blob/master/platform.txt#L100):

- the size of the code (segments `.irom0.text`, `.text`, `.data` and `.rodata`) 
- the size of global variables (segments `.data`,`.rodata` and `.bss`)

It **cannot** know the space occupied on the stack as it's constantly changing during the execution  of the program.

Did you try to use `DynamicJsonBuffer(1000)`?

MHz000, 2016-09-26:
Thank you for replaying. Indeed I've tried DynamicJsonBuffer. I'm totally confused now and put the project for a moment aside ...
Will come back later.

bblanchon, 2016-10-10:
@MHz000, any progress ?

MHz000, 2016-10-10:
Hi Benoit,
unfortunately no progress and no idea how to proceed  :-((  still getting

Exception (9):
epc1=0x40202846 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8542
depc=0x00000000

ctx: cont
sp: 3ffefa00 end: 3ffefc10 offset: 01a0

>>>stack>>>
3ffefba0:  3ffeeab0 3ffee4b0 0000003a 40202866
3ffefbb0:  3fffdad0 3ffee4b0 3ffeebb8 40201da5
3ffefbc0:  00000000 00000000 00000000 40202c78
3ffefbd0:  00000000 feefeffe 3ffeebb8 40202c9c
3ffefbe0:  00000000 00000000 00000001 3ffeebe4
3ffefbf0:  3fffdad0 00000000 3ffeebdc 40203150
3ffefc00:  feefeffe feefeffe 3ffeebf0 40100718
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

after one or two successfull requests.

It woud be fine if someone could make a little test on his system in his
envireoment to see it something is wrong in my setup.
Here is the simplified code
//--------------------------------------------------------------------------------------------------------------------------------------------------------------------
char versions_id[] = "ESP__WEB_test_V6";
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ArduinoJson.h>
#include <TimeLib.h>

const char WIFI_SSID[] = "...";               // WiFi information
const char WIFI_PSK[] = "...";

const char http_site[] = "api.usno.navy.mil";       // Remote site
information
const int http_port = 80;

const int LED_PIN = 16;                                   // global
variables & definitions
WiFiClient client;
char c;                                                              //
incomming char from WEB-server
unsigned long delayMillis;                               // delay loop time
unsigned long pause = 20000;                        // prog paused

// JSON stuff
StaticJsonBuffer<1000> jsonBuffer;                  // Memory pool for JSON
object tree.
char json[500];                                     // JSON object char
array, filld with server response
String head = "";                                   // HTTP header from
server
int json_i=0;                                       // index for json char
array
int JS_bracket = 0;                                 // counter for JSON {
and } brackets

void setup() {
  Serial.begin(115200);                             // Set up serial
console to read web page
  Serial.println(); Serial.println(versions_id);
  Serial.println("********************RE-/START******************");
  pinMode(LED_PIN, OUTPUT);                         // Set up LED for
debugging
}

void loop() {
  head = "";                                        // reset server
response string
  JS_bracket = 0;                                   // reset json bracket
counter

  connectWiFi();                                    // Connect to WiFi

  if ( !getPage() ) {                               // Attempt to connect
to website
    Serial.println("GET request failed");}
    else {
    Serial.println("GET request ok");}

    while (client.connected() ) {
    if ( client.available() ) {                     // If there are
incoming bytes, print them
      c = client.read();
      //Serial.print(c);
      if (JS_bracket == 0) {head=head+c;}           // fill in server
response
      if (c == '{') {++JS_bracket;}                 // ectraxt json object
      if (c == '}') {--JS_bracket; }
      // still to do: skip white space outside ".."
      if (JS_bracket > 0) { //Serial.print(c);      // look for matching
closing bracket
      json[json_i]=c; json_i++;}; }
    }
    json[json_i]='}';                               // add a closing }
bracket
    Serial.println("stop client");                  // If the server has
disconnected, stop the client and WiFi
    client.stop();                                  // Close socket and
wait for disconnect from WiFi
    if ( WiFi.status() != WL_DISCONNECTED ) {
      WiFi.disconnect();}
    digitalWrite(LED_PIN, HIGH);                    // Turn off LED

// *****process server data here
  Serial.println("processing server data");
//  for (int i=0; i<=json_i; i++) {Serial.print(json[i]);}
Serial.println();  // control print
  Serial.println("******************pausing****************");
  delayMillis= millis()+pause;                             // 5 minutes
intervall
   do {
     yield();
     } while (millis() < delayMillis);
  Serial.println("***********start after pausing***********");
// *****end processing server data

}

//******************connect WiFi***********************
void connectWiFi() {                            // connect WiFi
byte led_status = 0;
WiFi.mode(WIFI_STA);                            // Set WiFi mode to station
(client)
WiFi.begin(WIFI_SSID, WIFI_PSK);                // Initiate connection with
SSID and PSK
while ( WiFi.status() != WL_CONNECTED ) {       // Blink LED while we wait
for WiFi connection
  yield();
  digitalWrite(LED_PIN, led_status);
  led_status ^= 0x01;
  delay(100);
  }
digitalWrite(LED_PIN, HIGH);                    // Turn LED on when we are
connected
}

//***********************get page***************
bool getPage() {                                // Perform an HTTP GET
request to a remote page
if ( !client.connect(http_site, http_port) ) {  // Attempt to make a
connection to the remote server
return false;
}
client.println("GET /rstt/oneday?date=9/14/2016&coords=47.73N,9.16E&tz=1
HTTP/1.1"); // Make an HTTP GET request
client.print("Host: ");
client.println(http_site);
client.println("Connection: close");
client.println();
return true;
}
//------------------------------------------------------------------------------------------------------------------------
enviroment:  WIN10 arduion 1.6.12

Thank you for any help or hint

bblanchon, 2016-10-10:
@MHz000 `jsonBuffer` is declared but isn't referenced anywhere. did forget something?

Also, I cannot repeat this enough: it's not possible to use a global `JsonBuffer`.

See:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-shouldnt-i-use-a-global-jsonbuffer)

MHz000, 2016-10-11:
Hi Benoit,
thank you for your patience and your hint. I had stripped down my sketch
for debugging too much :-(

Regards

bblanchon, 2016-10-16:
@MHz000, any progress?

MHz000, 2016-10-16:
yes indeed. Didn't I report it? Sorry. As you mentioned it: keep
*json local! *
thanks again for patinetce and help

bblanchon, 2016-10-16:
You're welcome @MHz000, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-10-16 by bblanchon