bblanchon/ArduinoJson #360: Nested JSON buffer print to string

sachinranka4u, 2016-09-29:
Hi
I am back with another query, pls bear with me if I am asking silly doubts! ;)
I am really becoming your fan as I have started using your libraries!...

I am trying to create a nested Array inside a JSON object,

CODE used:
`StaticJsonBuffer<200> jsonBuffer;`
`JsonObject& jsonObj = jsonBuffer.createObject();`
`JsonArray& p1 = jsonObj.createNestedArray("p1");`
`void setup(){`
`    ...`
`}`
`void loop(){`
`...`
`  p1.add("0");`
`  p1.add("1");`
`  ....`
`  jsonObj.printTo(Console);`
`  char strObj[156];`
`  jsonObj.printTo(strObj, sizeof(strObj));`
`  ...`
`}`

But when I try to print it I get ouput as only
`{"p1":[]}` instead of `{"p1":[0, 1]}` 

Is there somewhere I am missing out something?

bblanchon, 2016-09-29:
Hi Sachin,

Your `JsonBuffer` is probably too small, you need to increase the size.
Also, using a global `JsonBuffer` is asking for trouble.

See:

* [FAQ: Why some parts are missing?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-some-parts-are-missing)
* [FAQ: The first serialization succeeds, why do the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-serialization-succeeds-why-do-the-next-ones-fail)
* [FAQ: Why shouldn't I use a global JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-shouldnt-i-use-a-global-jsonbuffer)

sachinranka4u, 2016-09-29:
Wow :+1: , I made it as a `DynamicJsonBuffer` instead of static and it worked!
Also used then inside the loop function where it would be required and introduced my own dataset n _Woila!_ It worked  :100:  :)
It would be a nice point to say that this library should be used for serialization rather than assuming this as a datasource, just a suggestion :)

closed 2016-09-29 by sachinranka4u

bblanchon, 2016-09-29:
You're welcome Sachin, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: