bblanchon/ArduinoJson #361: Using the library outside the Arduino

leventesen, 2016-10-01:
Is it possible to port and use the ArduinoJson library without the Arduino, for example in AVR MCU - GNU Toolchain - Atmel Studio?

What are the Arduino dependencies to remove and which compiler/linker options must be set?

bblanchon, 2016-10-01:
ArduinoJson is not dependent on Arduino, nor anything else.

It's a header-only library, so you just have to `#include <ArduinoJson.h>` and that's all.

You do not need any compiler/linker flags.
However you'll have a warning if you use the flag `--fpack-struct`, you'll have the following warning:

    ignoring packed attribute because of unpacked non-POD field

See issue #255.

leventesen, 2016-10-01:
When I build, compiler gives this error:

undefined reference to __cxa_guard_acquire
undefined reference to __cxa_guard_release

bblanchon, 2016-10-01:
This should be fixed by adding the flags `-fno-threadsafe-statics`.

See issue #356.

closed 2016-10-01 by leventesen