bblanchon/ArduinoJson #362: Adding nested objects in a loop

koffienl, 2016-10-04:


bblanchon, 2016-10-05:
ArduinoJson has a different behavior with `String`: it copies its content into the `JsonBuffer`.
This is based on the assumption that most people use `String` as a temporary value, just like you.
In your code, this behavior was not triggered because you called `String::c_str()`.

Just replace:

```c++
for (uint8_t i = 0; i < 5; i++)
  {
    String Step = "mash" + i;
    if (!mash.containsKey(const_cast<char*>(Step.c_str())))
    {
      mash.createNestedObject(const_cast<char*>(Step.c_str()));
    }
  }
```

by

```c++
for (uint8_t i = 0; i < 5; i++)
  {
    String Step = "mash" + i;
    if (!mash.containsKey(Step))
    {
      mash.createNestedObject(Step);
    }
  }
```

koffienl, 2016-10-05:
Got it working, thanks for pointing the problem :)

bblanchon, 2016-10-05:
You're welcome @koffienl, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-10-05 by bblanchon