bblanchon/ArduinoJson #363: Decoding encoded JSON that does not feature escape sequences

schirag1993, 2016-10-06:
I'm currently working on a multi-Arduino project. Here, sensor data is encoded perfectly and sent to another Arduino. Communication occurs through the NRF24L01 module. While information is being sent successfully, the receiving Arduino is unable to decipher the message. I believe this is due to the absence of escape sequences.

Here is a sample JSON that I'm trying to send:
**{"sensor":"mpu","xaxis":"r","yaxis":"f"}**

This is the exact message that is being sent to the other Arduino. No output whatsoever results with the usage of the conventional radio.read() to receive and store data into a char[]. Attempted the usage of a structure. However, the resulting erratic breakup of key/value pairs is a little cumbersome to process further. 

Output of erratic breakup of JSON when using structures here:
**{"sensor":"mpu","xaxis"
"xaxis":"halt"
,"**

The decoding example in the examples section showcases the decoding of a char array with escape sequences already in place. I was wondering whether it is possible directly decode the JSON string as is. If not, any other ways that the JSON string can be designed to allow for easy deciphering?

My apologies if this question seems mundane. I have only recently entered the realm of programming and IoT.

bblanchon, 2016-10-06:
This reminds me of #327.
Can you please check if it's the same problem?

schirag1993, 2016-10-07:
I did check #327, in that issue, the problem was that the values within the JSON itself was changing. Here, the values do not change. It is just that I am unable to receive the transmitted JSON in the form of a character array on the other side. I believe this is due to the absence of escape sequences. 

I believe the confusion has risen from the fact that I have posted alternate key/value pairs when showcasing the erratic behavior of the structure method. My apologies for that oversight. I was fiddling with the code between the time I recorded the input and the output. We can safely assume that the key/value pairs do not change at all. 

The only problem I am facing is receiving the JSON string in a consumable format on the other Arduino. Is there a way I can add the escape sequences through the use of your library? Or must it be done manually?

schirag1993, 2016-10-07:
Hi, it seems that the entire JSON string can be stored in a character array provided the JSON string is small enough. Previously, the JSON string was 40 characters long. With a reduction to 31 characters, the string was successfully transmitted. In light of this new knowledge, I believe the main culprit was packet loss during transmission and NOT some strange issue with escape sequences. 
However, I did take precautions against packet loss by following a few suggestions online. Although the issue seems to be resolved, I'd still like an opinion from a more experienced individual such as yourself. Please do let me know if you need anything I may have omitted (such as code, output screenshots, setup structure etc.)

bblanchon, 2016-10-07:
Hi,

I'm not sure how I can help you, as the problem is not related to ArduinoJson.
If you want I can do a review of a **small** code snippet using ArduinoJson, to make sure everything is correct on this side.

schirag1993, 2016-10-07:
Yes, I understand. Your library works flawlessly, no doubt about that. As this is an issue with packet loss, I will take leave of this conversation. Thank you for your time Mr. Benoît Blanchon. 

And thank you for the library! :)

closed 2016-10-07 by schirag1993

bblanchon, 2016-10-07:
You're welcome.
Thank you for using ArduinoJson.