bblanchon/ArduinoJson #364: Unable to add to existing array

devinrayolsen, 2016-10-07:
I'm interfacing with an SD card to write and read settings from files through simple get and set functions. So far I've had no problem with changing values and removing arrays from the data read from a SD file and save it back out to the file. What I'm not able to do however is add to object via set, or add.

Here are my get and set functions:
```c++
JsonObject& getCropData(){
	DynamicJsonBuffer b;
	tmpFile = SD.open("dromatic/"+ cropName + "/crop.dro");
	JsonObject& d = b.parseObject(tmpFile.readString());
	tmpFile.close();
	return d;
}
void setCropData(JsonObject& d){
	char b[512];
	tmpFile = SD.open("dromatic/" + cropName + "/crop.dro", O_WRITE | O_TRUNC);
	d.printTo(b, sizeof(b));
	tmpFile.print(b);
	tmpFile.close();
}
```

The crop file look like this:
```json
{
	"ec":[1200,1600],
	"ph":[5.60,6.20],
	"phChannels":[1,2],
	"sessions":[
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2]
	],
	"totalChannels":10
}
```

As stated above, I'm able to remove elements from the nested sessions array:
```json
{
	"ec":[1200,1600],
	"ph":[5.60,6.20],
	"phChannels":[1,2],
	"sessions":[
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2],
		[2016,10,6,4,11,12,0,0,0,2]
	],
	"totalChannels":5
}
```

But not able to add more entries to sessions.
Here is a snip from the function that adds/removes session entries:
```c++
int i;
JsonObject& cropData = getCropData();
JsonArray& cropSessions = cropData["sessions"].asArray();
Time currentTime = rtc.getTime();
for (i = 1; i < 10; i++){
	//dir = direction in which we are altering the sessions array (trim or add).
	if (dir < 0 && i >= newTotal){ //triming
		cropSessions.removeAt(i);
	}
	if (dir > 0 && i > currentTotal){ //adding
		JsonArray& newSession = cropSessions[i].asArray();
		newSession[0] = currentTime.year;
		newSession[1] = currentTime.mon;
		newSession[2] = currentTime.date;
		newSession[3] = currentTime.dow;
		newSession[4] = currentTime.hour;
		newSession[5] = currentTime.min;
		newSession[6] = 0;
		newSession[7] = 0;
		newSession[8] = 0;
		newSession[9] = 2;
	}
}
```
I know I'm missing something obvious just not sure what I'm doing wrong.
Thanks!

bblanchon, 2016-10-07:
This is incorrect:

```c++
JsonObject& getCropData(){
    DynamicJsonBuffer b;
    JsonObject& d = b.parseObject();

    return d;
}
```

because you return a reference to a `JsonObject` stored in the `JsonBuffer b` and therefore is discarded as the function returns.

Here is a workaround:

```c++
JsonObject& getCropData(JsonBuffer& b){
    JsonObject& d = b.parseObject();

    return d;
}
```

devinrayolsen, 2016-10-07:
Success, thanks man!
I know that handing off the buffer to the function was an option, but my original function still allowed me to read, change and remove existing values/elements just not add, is that normal behavior?

bblanchon, 2016-10-07:
Yes, this is what C++ devs call "undefined behavior" or UB.
It can work for a little time, but it's very likely to fail in an unexpected way.

closed 2016-10-07 by bblanchon