bblanchon/ArduinoJson #365: Problem parsing large datasets in multi-dimensional arrays

cvrdx, 2016-10-07:
Hello,
First of all thanks for this awesome library that is really helpful! However I'm new to Json in general and I have a problem... Could you help me?

Basically I need to parse some data which looks like this: char jdataSet1[] = "{\"chn\":[1,2,3],\"data\":[[[255,52,13],[255,187,0],[255,129,133],[255,18,0]],[[251,107,38],[255,192,0],[255,135,5],[250,132,64]],[[253,138,91],[254,248,212],[255,33,0],[196,187,1]],[[202,249,221],[236,127,0],[255,6,143],[245,255,133]]]}";
I have 4 of these.

I want to parse them into multi-dimensional arrays like this: int dataSet1[PresetCount][SubPresetCount][3];

I'm running it on an Arduino Uno (Atmega 328p with 32kbytes of Flash and 2kbytes of Ram). It's not a lot so I need to be as efficient as possible to leave room for other functions as well.

I've tried different things but I couldn't parse all my data while being memory efficient. I got to the point where I adapted the HTTPClient example and it works, but only when I parse 1 dataset. If I want to parse more (by uncommenting the function calls in the defineData function), the Serial Monitor freezes. I've tried increasing the JsonBuffer size but it doesn't have any effect. Plus, the size of each dataset is <700 according to the calculator. And this already uses 48% of the Ram.
Here's the code:

```
#include <ArduinoJson.h>

#define PresetCount 4
#define SubPresetCount 4 
int dataSet1[PresetCount][SubPresetCount][3];
int dataSet2[PresetCount][SubPresetCount][3];
int dataSet3[PresetCount][SubPresetCount][3];
int dataSet4[PresetCount][SubPresetCount][3];
int dataSetChnl1[3];
int dataSetChnl2[3];
int dataSetChnl3[3];
int dataSetChnl4[3];

void setup() {
Serial.begin(9600);
defineData();
}

void loop() {
// Nothing here
}

void defineData() 
{
char jdataSet1[] = "{\"chn\":[1,2,3],\"data\":[[[255,52,13],[255,187,0],[255,129,133],[255,18,0]],[[251,107,38],[255,192,0],[255,135,5],[250,132,64]],[[253,138,91],[254,248,212],[255,33,0],[196,187,1]],[[202,249,221],[236,127,0],[255,6,143],[245,255,133]]]}";
char jdataSet2[] = "{\"chn\":[4,5,6],\"data\":[[[0,208,93],[255,57,84],[253,102,0],[220,1,13]],[[255,238,0],[0,255,145],[66,154,217],[220,212,0]],[[214,34,28],[255,202,37],[0,140,255],[299,59,0]],[[249,191,5],[199,245,242],[255,124,6],[0,37,255]]]}";
char jdataSet3[] = "{\"chn\":[7,8,9],\"data\":[[[204,0,0],[255,204,219],[254,5,6],[249,200,0]],[[148,255,112],[255,177,0],[255,60,0],[253,90,25]],[[238,255,0],[229,98,0],[253,88,7],[253,177,55]],[[255,130,1],[233,165,2],[223,170,180],[0,190,98]]]}";
char jdataSet4[] = "{\"chn\":[10,11,12],\"data\":[[[255,34,0],[73,253,28],[255,206,51],[255,58,1]],[[255,134,0],[255,125,0],[255,195,1],[255,219,10]],[[255,5,4],[239,211,0],[187,0,5],[250,229,124]],[[255,170,0],[245,7,4],[246,7,28],[0,237,249]]]}";

//bool test1 = 
parseUserData(jdataSet1, 1);
parseUserData(jdataSet2, 2);
//bool test2 = parseUserData(jdataSet2, 2);
//bool test3 = parseUserData(jdataSet3, 3);
//bool test4 = parseUserData(jdataSet4, 4);
}

void parseUserData(char* content, int selection) {

  StaticJsonBuffer<2000> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(content);

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
  //  return false;
  } Serial.println("parsing OK");
 // return true;
  
  switch (selection) {
      case 1:
        for (int i = 0; i < PresetCount; i++) {
          for (int y = 0; y < SubPresetCount; y++) {
            for (int z = 0; z < 3; z++) {
              dataSet1[i][y][z] = root["data"][i][y][z];
            }
          }
          if (i < 3) {
            dataSetChnl1[i] = root["chn"][i] ;
          }
        }
Serial.println("Parsing Dataset 1 Done");
        break;
        
      case 2:
        for (int i = 0; i < PresetCount; i++) {
          for (int y = 0; y < SubPresetCount; y++) {
            for (int z = 0; z < 3; z++) {
              dataSet2[i][y][z] = root["data"][i][y][z];
            }
          }
          if (i < 3) {
            dataSetChnl2[i] = root["chn"][i] ;
          }
        }
Serial.println("Parsing Dataset 2 Done");
        break; 
    }
}

```
Do you know what's wrong? Thanks very much!
Best

Cor

bblanchon, 2016-10-07:
Hello Cor,

You don't have enough memory to do that.

`StaticJsonBuffer<2000>` on a device that has 2KB of RAM.
This is simply impossible.

Even if you reduce the size of the `StaticJsonBuffer`, all your variables are eating a most of the RAM: `dataSetX`, `jdataSetX` etc.

You need to reduce the number of variables in memory, or buy another MCU.

Regards,
Benoit

cvrdx, 2016-10-13:
Hi Benoit
Thanks for your answer. Indeed I only managed to parse one dataset at a time with a `StaticJsonBuffer `size of 700, as more made my microcontroller freeze. I found another way to store my data using the internal EEPROM.
Regards
Cor

closed 2016-10-13 by cvrdx

bblanchon, 2016-10-13:
You're welcome Cor.
Thank you for using ArduinoJson.