bblanchon/ArduinoJson #366: Weird symbols for JsonObject

fabiosoft, 2016-10-08:
I encode the json like this: 

      DynamicJsonBuffer jsonBuffer;
      JsonObject& ack = jsonBuffer.createObject();
      ack[F("id")] = Discovery::getChipId_hex();
      ack[F("ssid")] = WiFi.SSID();
      ack[F("localip")] = WiFi.localIP().toString();
      ack[F("mac")] = WiFi.macAddress();

      int len = ack.measureLength();
      char buffer[len];
      ack.printTo(DEBUG_SERIAL);
      ack.printTo(buffer, ack.measureLength());

But instead to get printed this:

    {"id": "xxxxx","ssid": "fabio_link_fast","localip": "192.168.1.106","mac": "XX:XX:XX:XX:XX:XX"}

I get

    {"\bd":"\4\c3\34\4","ss\bd":"\c\fb\bo_l\bnk_\c\fst","lo\4\fl\bp":"1\32.168.1.106","m\f\4":"XX:XX:\3:XX"

What is this?? **How to solve?**

bblanchon, 2016-10-09:
Even if that's not the cause of your problem, I highly recommend not to use flash strings with ArduinoJson (at least not now, see #76).

I tried your code and it works.
Did you show all the relevant code?

See:

* [FAQ: Why does the generated JSON contain garbage?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-does-the-generated-json-contain-garbage)

bblanchon, 2016-10-14:
@fabiosoft, any progress?

closed 2016-10-23 by bblanchon