bblanchon/ArduinoJson #368: JsonObjectKey doesn't copy const char *

anisimovsergey, 2016-10-09:


bblanchon, 2016-10-09:
Yes, by default strings are not copied, and that's by design.
It allows a zero-alloc/zero-copy serialization, that's the whole point of the library.

However, `std::string`s are copied when needed.
Just change your function like this:

```c++
void Foo:setValue(const std::string& key, const std::string& value) {
  jsonObject[key] = value;
}
```

@anisimovsergey, I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

anisimovsergey, 2016-10-09:
The documentation is good, it's just me. I read it half a year ago and forgot about this nuance. 

The solution proposed gives the following error:
.piolibdeps\ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:62:15: note:   no known conversion for argument 1 from 'const string {aka const std::basic_string<char>}' to 'ArduinoJson::JsonObjectKey'

But don't worry, I'll find something out.

bblanchon, 2016-10-09:
What compiler are you using?

anisimovsergey, 2016-10-09:
GCC 4.8.2 (espressif8266 platform on Platform.io)

bblanchon, 2016-10-10:
[ArduinoJson works either with Arduino's `String` or `std::string`](https://github.com/bblanchon/ArduinoJson/blob/v5.6.7/include/ArduinoJson/String.hpp#L21)

If you want to force `std::string`, you need to define `ARDUINOJSON_USE_ARDUINO_STRING` to `0`.

anisimovsergey, 2016-10-10:
Ah, awesome, exactly what I needed.

bblanchon, 2016-10-10:
You're welcome Sergey, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-10-10 by bblanchon