bblanchon/ArduinoJson #370: Compiler crash when add EasyTransfer library

choncba, 2016-10-11:
Hello everyone, first of all I want to congratulate the creator of this excellent library.
I use a ESP8266 ESP12 , in one of my projects I use EasyTransfer library (https://github.com/madsci1016/Arduino-EasyTransfer) to synchronize with another arduino via serial link.
When trying to compile ArduinoJSON with this compiler fails with the following message:

ArduinoJson.hpp:11: In file included from
ArduinoJson.h:8: from
ArduinoJson.h:8: from
RiegoControlV4ESP.ino:47: from
JsonArray.hpp: In instantiation of typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonArray::get(size_t) const [with T = char; typename ArduinoJson::Internals::JsonVariantAs<T>::type = char; size_t = unsigned int]
JsonArraySubscript.hpp:55: required from typename ArduinoJson  Internals  JsonVariantAs<T>  type ArduinoJson  JsonArraySubscript  as() const [with T = char; typename ArduinoJson  Internals  JsonVariantAs<T>  type = char]
JsonVariantBase.hpp:61: required from const typename ArduinoJson  Internals  JsonVariantAs<T>  type ArduinoJson  JsonVariantBase<TImpl>  as() const [with T = char; TImpl = ArduinoJson  JsonArraySubscript; typename ArduinoJson  Internals  JsonVariantAs<T>  type = char]
JsonVariantBase.hpp:56: required from ArduinoJson  JsonVariantBase<TImpl>  operator T() const [with T = char; TImpl = ArduinoJson  JsonArraySubscript]
RiegoControlV4ESP.ino:583: required from here
 
JsonArray.hpp: 142:17: error: no matching function for call to 'ArduinoJson::JsonVariant::as()
   JsonVariant::defaultValue<T>()
JsonArray.hpp:142: note  candidates are

This library uses malloc () to allocate memory.

Any ideas? 

Thanks.-

bblanchon, 2016-10-11:
Hello Luciano,

It seems that you are trying to convert a `JsonVariant` to `char`, which is not supported.

You probably have code like this:

```c++
char c = myObject["key"];
```

you just need to change it to: 

```c++
byte c = myObject["key"];
```

See:

* [FAQ: Why JsonVariant cannot be converted to char?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-jsonvariant-cannot-be-converted-to-char)

choncba, 2016-10-11:
Thank you @bblanchon !
I was focused on thinking it was a problem of memory allocation and did not see the obvious.
Here is my code:

StaticJsonBuffer<BufferSize> buffer;
JsonArray& recibido = jsonBuffer.parseArray(server.arg("plain"));

```
if (recibido.success())
{
	int NumSalida = recibido[0];
	Estados.EstadoSalida[NumSalida] = recibido[1];
	// SendCmd() takes 3 chars as argument
        //SendCmd('S', recibido[1], recibido[0]);	// -> This way produces the error
        SendCmd('S', Estados.EstadoSalida[NumSalida], NumSalida);	// This way is ok
}

```
So, the EasyTransfer library has nothing to see with this error.

closed 2016-10-11 by choncba

bblanchon, 2016-10-11:
You're welcome Luciano, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: