bblanchon/ArduinoJson #371: OTA compatibility

odilonafonso, 2016-10-12:
### Basic Infos

#### Hardware
Hardware:		ESP8266 NodeMCU
Core Version:      	2.3.0-rc2

### Description

Sketch crash when try to use ArduinoJson with OTA httpUpdate.

I'm trying to use the library ArduinoJson with an already functional code firmware update via OTA based httpUpdate.

Simply place the instructions:

>     StaticJsonBuffer <200> jsonBuffer;
>     JSONObject & httpJson jsonBuffer.createObject = ();
> 

so that the sketch to crash.

There are some marked incompatibility that I have not seen?

bblanchon, 2016-10-13:
@odilonafonso, I'd glad to help you but, unfortunately, I don't understand your question.
Can you please give more (but not too much) of context?

odilonafonso, 2016-10-13:
Hi @bblanchon,

Thanks for help !

I used the httpUpdate sample from ESP8266httpUpdate library, for test how to implement a OTA update on my applications. 

With minor modifications, the tests was good, very good.

Now, I want to use your excellent library for implement a little API, sending to my application something like:

update={"version":"v1.0 bla bla bla","server":"http://myserver.com","binfile":"/bin/myESPapp.bin"}

Well, the first test was frustrating, because I can not load the sketch only with a call to the ArduinoJson library.

Now, I have a working version, but I do not understand why did not worked first time. And it seems to me that the environment is very sensitive to changes.

Please see the sketch above; It is working, but I just make some small changes and it stops working.

```
/**
 * httpUpdate.ino
 *
 *  Created on: 27.11.2015
 *
 */
#include "ESP8266WiFiMulti.h"
#include "ESP8266httpUpdate.h"
#include "ArduinoJson.h"

#define SSID_PONESA "MY_SSID1"
#define PASSWORD_PONESA "MY_PASS1"
#define SSID_ALBACORA "MY_SSID2"
#define PASSWORD_ALBACORA "MY_PASS2"

ESP8266WiFiMulti WiFiMulti;

#define DEBUG_ESP_PORT Serial
#define DEBUG ESP_HTTP_UPDATE

#define LOCAL_SERVER "http://odilon.local"
#define REMOTE_SERVER "http://robuino.com.br"
#define SKETCH_BIN "/httpUpdate.ino.bin"
#define THIS_VERSION "V0.3(no boot after update)"
#define BOOT_AFTER_UPDATE false
int updateCount = 1;

/*
 * gerar os dados em Json - para testar a forma como ficará em produção
 * a string gerada será da forma:
 * {"version":"v0.2xxx","server":"http://robuino.com","binfile":"/robuino_v0.2.bin"}
 * 
 * Step 1: Reserve memory space
 */

// I HAVE TO USE StaticJsonBuffer global; 
// DO NOT WORK (CRASH) WHEN DECLARED INSIDE updateBin() function
StaticJsonBuffer<300> jsonBuffer;

void updateBin() {
   Serial.print("\nhttpUpdate");
   Serial.print(THIS_VERSION);
   Serial.print("loop...");
   Serial.println(updateCount++);

/*
 * Step 2: Build object tree in memory
 */
   JsonObject& httpJson = jsonBuffer.createObject();
   httpJson["version"] = THIS_VERSION; //"V0.3(no boot after update)"; //
   httpJson["server"] = REMOTE_SERVER; //"http://robuino.com.br"; //
   httpJson["binfile"] = SKETCH_BIN; //"/httpUpdate.ino.bin"; //
//
   String httpString;
   httpJson.printTo(Serial);
//   httpJson.printTo(httpString); THIS NOT WORK !!! Sketch crash on boot, no messages...
//   Serial.println(httpString);

   String url = httpJson["server"]; //REMOTE_SERVER; //
   String bin = httpJson["binfile"]; //SKETCH_BIN; //
   url += bin;
   Serial.print("url: ");Serial.println(url);
//this method do not work...
//      t_httpUpdate_return ret = ESPhttpUpdate.update(REMOTE_SERVER, 80, SKETCH_BIN);
   ESPhttpUpdate.rebootOnUpdate(BOOT_AFTER_UPDATE);
   t_httpUpdate_return ret = ESPhttpUpdate.update(url);
   Serial.print("ret ");Serial.println(ret);

   switch(ret) {
      case HTTP_UPDATE_FAILED:
          Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
          break;
      
      case HTTP_UPDATE_NO_UPDATES:
          Serial.println("HTTP_UPDATE_NO_UPDATES");
          break;
      
      case HTTP_UPDATE_OK:
          Serial.println("HTTP_UPDATE_OK");
          break;
      
      default:
          Serial.print("Undefined HTTP_UPDATE Code: ");Serial.println(ret);
   }
}

void setup() {
   Serial.begin(115200);
   Serial.setDebugOutput(true);

   Serial.println();
   Serial.println();
   Serial.println();

   for(uint8_t t = 4; t > 0; t--) {
     Serial.printf("[SETUP] WAIT %d...\n", t);
     Serial.flush();
     delay(500);
   }

   Serial.print("\nhttpUpdate");
   Serial.print(THIS_VERSION);
   Serial.println("start...");

   WiFiMulti.addAP(SSID_PONESA, PASSWORD_PONESA);
   WiFiMulti.addAP(SSID_ALBACORA, PASSWORD_ALBACORA);

}

void loop() {
   // wait for WiFi connection
   if((WiFiMulti.run() == WL_CONNECTED)) {
      updateBin();
   }
   delay(1000);
}

```

I noticed a strange behavior, for example to change the order in which libraries are called in the sketch, the sketch hangs. For example, I can not call ArduinoJson library before others.

Thank you very much your help, I will make some more tests and post here what you can observe.

bblanchon, 2016-10-13:
```
// I HAVE TO USE StaticJsonBuffer global; 
// DO NOT WORK (CRASH) WHEN DECLARED INSIDE updateBin() function
```

Interesting... did you try with `DynamicJsonBuffer`?

odilonafonso, 2016-10-13:
No; the way it is now is working. I'll do some more tests, trying to figure out something else.
Grateful !

closed 2016-10-14 by bblanchon