bblanchon/ArduinoJson #372: Using add operator in Json & start index from zero

vindhyachaltakniki, 2016-10-14:
1. I have defined Json using:
StaticJsonBuffer<80> jsonBuffer;
JsonArray& array = jsonBuffer.createArray();

2. Now I keep on using add operator like as below & keep on adding values in array like this: 
        array.add("Value");
        array.add(data[0]);

3. After one loop, I want to reinit the Json array index to 0 again. So that when I now add values , it should again start from index 0. How to do that?

4. Second question is how to know how much array size is filled in Json array at any point of time. This will help in case, intiially I will define a large array, now when I keep on adding values in it, in the end I want to know whats its size know. So that I can optimize the size of initially defined Json array.

bblanchon, 2016-10-14:
You can set the value of an element of an array like that:

```
array[0] = "Value";
array[1] = data[0];
```

You can get the number of elements in the array by calling:

```
int number_of_elements = array.size();
```

See:

* [API Reference: JsonArray::operator[]](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference#jsonarrayoperator)
* [API Reference: JsonArray::set()](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference#jsonarrayset)
* [API Reference: JsonArray::size()](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference#jsonarraysize)

vindhyachaltakniki, 2016-10-15:
Hi,

I redefine the JSON array everytime in a loop, as in below code.  Problem is if my code I use:

1.  "StaticJsonBuffer<500> jsonBuffer;",
Code gives correct output & length of arry is 217 as printed.

["$",255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,"&"]

217

2. But if I use "StaticJsonBuffer<316> jsonBuffer;",
code gives incorrect output, even though max length of array is 17. Even now incorrect array length is printed: 157, which is wrong. 
In the end "&" should be added, but nows it dont add it the end. Now output is:

["$",255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255]

157

`#include <ArduinoJson.h>

void setup() 
{
/* serial 0 init */
    Serial.begin(9600);
}

void loop()
{
    uint8_t cnt;
    uint8_t em6433_arr[51];
    uint8_t slave = 255U;

    for(cnt = 0U ; cnt < 51U ; cnt++)
    {
        em6433_arr[cnt] = 255;
    }

    /* JSON */
    StaticJsonBuffer<316> jsonBuffer;
    JsonArray& array = jsonBuffer.createArray();
      

    array.add("$");
    array.add(slave);  /* slave id */
    for(cnt = 0U ; cnt < 51U ; cnt++)
    {
      array.add(em6433_arr[cnt]);
    }
    array.add("&");
    array.printTo(Serial);
    Serial.println("\n");
    Serial.println(array.measureLength());
    Serial.println("\n");
 

    delay(2000);
              
   //while(1);
}`

bblanchon, 2016-10-15:
The [JsonBuffer size calculator](https://rawgit.com/bblanchon/ArduinoJson/master/scripts/buffer-size-calculator.html) says you need 436 bytes, so it normal that it works with 500 and it's truncated with 316.

vindhyachaltakniki, 2016-10-21:
But on using array.measureLength(), it returns 217 & this is the number of in array then how comes this calculator shows 436 bytes??

bblanchon, 2016-10-21:
These are two different things:

* `measureLength()` returns the length of the JSON string
* the calculator computes the size of the `JsonBuffer`

See:

* [ArduinoJson: Memory Model](https://github.com/bblanchon/ArduinoJson/wiki/Memory%20model)

bblanchon, 2016-10-25:
@vindhyachaltakniki Can I close this issue?

vindhyachaltakniki, 2016-10-25:
yes, thanks for help

bblanchon, 2016-10-25:
You're welcome @vindhyachaltakniki, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-10-25 by bblanchon