bblanchon/ArduinoJson #373: How to store generated JSON in a String

Pooja-B, 2016-10-14:
Hi,
I want to send the JSON data to cloud, JSON generation is successful and it is printing to serial aswell but when I am trying to store this generated JSON into a string or Char array it is not working. Please let me know  a solution for this, so I can store JSON and perform further operation on it.

Regards
Pooja

bblanchon, 2016-10-14:
Hi,

```c++
// to generate a char[]:
char json[256];
root.printTo(json, sizeof(json));

// to generate a String
String json;
root.printTo(json);
```

Pooja, I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

Pooja-B, 2016-10-17:
Hi Benoît, 
Thanks for the solution. Actually I didn't went through the complete documentation. Thanks for creating and maintaining this repository so well. The documentation is really nice and helpful and there is almost everything one needs to work with JSON and arduino. I have also created few good examples with this library and ESP8266, I will publish that soon.

Regards
Pooja

bblanchon, 2016-10-17:
You're welcome Pooja, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

closed 2016-10-17 by bblanchon