bblanchon/ArduinoJson #374: Trouble adding string to array

arlinjv, 2016-10-15:
I'm trying to create a JSON string that contains an array with two string elements. It should look like this: 
{"status":"online","url":"192.168.50.150","resources":["relay","led"]}.

But I get this:
{"status":"online","url":"192.168.50.150","resources":["relay"]}

So the second 'add' doesn't seem to do anything.

Here's my code:

  StaticJsonBuffer<100> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["status"] = "online";
  root["url"] = WiFi.localIP().toString();
  JsonArray& resources = root.createNestedArray("resources");
  resources.add("relay");
  resources.add("led");
  char buff[100];
  root.printTo(buff,sizeof(buff));
client.publish("garden/master-irrigation", buff,true); 

I'm sure I'm doing something wrong, but I can't see what it might be.
This is running on a ESP8266. 

Thanks!

sticilface, 2016-10-16:
Use dynamic buffer. Not static on esp8266. Your allocation is too small anyway. 

Hope that helps.

bblanchon, 2016-10-16:
See:

* [FAQ: Why some parts are missing?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-some-parts-are-missing)

arlinjv, 2016-10-18:
Awesome. Easy fix. Thank you!

closed 2016-10-18 by arlinjv

bblanchon, 2016-10-18:
You're welcome @arlinjv, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: