bblanchon/ArduinoJson #376: Newbie question about fetching json values

MHz000, 2016-10-25:
Newbie question:
how to fetch "sundata":"phen":"BC", "time": values from this json object?
{
"error":false,
"apiversion":"1.3.0",
"year":2016,
"month":10,
"day":25,
"dayofweek":"Tuesday",
"datechanged":false,
"lon":9.160000,
"lat":47.730000,
"tz":1,

"sundata":[
            {"phen":"BC", "time":"06:26"},
            {"phen":"R", "time":"06:57"},
            {"phen":"U", "time":"12:07"},
            {"phen":"S", "time":"17:17"},
            {"phen":"EC", "time":"17:48"}],

"moondata":[
            {"phen":"R", "time":"01:21"},
            {"phen":"U", "time":"08:22"},
            {"phen":"S", "time":"15:14"}],

"closestphase":{"phase":"Last Quarter","date":"October 22, 2016","time":"20:14"},
"fracillum":"24%",
"curphase":"Waning Crescent"
}
Tried: String BC = root["sundata"]["phen"]["BC"]["time"];   however this failed. As I said I'm not so familiar with handling json's
Thank you

bblanchon, 2016-10-25:
`root["sundata"]` is an array, so it's indexed by an integer, not a string.

```c++
root["sundata"][0]["time"];
```

See:

* [FAQ: Parsing succeeds but I can't read the values!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#parsing-succeeds-but-i-cant-read-the-values)

MHz000, 2016-10-25:
Thank you so much Benoit!

bblanchon, 2016-10-25:
You're welcome.
Thank you for using ArduinoJson.

closed 2016-10-25 by bblanchon