bblanchon/ArduinoJson #377: json parser returns an error in some condition

vitasam, 2016-10-27:
Hello,
could you kindly point me where is a problem?

I want to add possibility to save configuration in to EEPROM after sending 'c' command through serial console. JSON parser is Ok only if code is stopped with while(1) loop, otherwise JSON parser returns an error (see code below).

I believe that there is something related to the serial buffer.

```
void setup()
{
    int counter = 7;
    bool get_config = false;

    // Open serial communications and wait for port to open:
    Serial.begin(SERIAL_PORT_SPEED);
    while (!Serial) ; // Wait for serial port to connect. Needed for native USB port only

    // Read configuration from the EEPROM
    EEPROM.get(0, module_config);
    print_config_to_serial(&module_config);

    Serial.println(">> Send 'c' to start JSON configuration or wait 7 sec to boot up");
    Serial.flush();
...
  while(counter > 0)
    {
        if(Serial.available() != 0)
        {
            int read_serial = Serial.read();
            if(read_serial == 'c')
            {
                Serial.println("\n>> Waiting for JSON configuration string");
                get_config = true;
                break;
            }
        }

        Serial.print(counter);
        Serial.print(" ");
        delay(1000);
        counter--;
    }

    if(get_config)
    {     
        StaticJsonBuffer<200> json_buffer;
        get_config = false;
        memset(&module_config, 0, sizeof(module_config));
  
        while(Serial.available() == 0){}    
        char char_buf[180];    

        String conf_string = Serial.readString();
        conf_string.toCharArray(char_buf, 180);  
              
        JsonObject& root = json_buffer.parseObject(char_buf);

        if (root.success())
        {
            module_config.module_ip_address[0] = root["mip"][0];
...
            print_config_to_serial(&module_config);
        }
        else
        {
            Serial.println(">> JSON parser error!");
            while(1){}           
        }
    }

    **// TODO: JSON works only if here while(1)
    //while(1){}**
...
```

bblanchon, 2016-10-27:
Hello,

The only thing that bothers me is that you use `readString()`, you should use [`readBytes()`](https://www.arduino.cc/en/Reference/StreamReadBytes) or [`readBytesUntil()`](https://www.arduino.cc/en/Reference/StreamReadBytesUntil).

I bet that the input is incorrect or the `StaticJsonBuffer` is too small.
Can you try to print the content of `char_buf` before calling `parseObject()`?

vitasam, 2016-10-27:
Hi,
apparently there is something wrong with char buffer. I following test:
1) hardcoded a test char array:
```
        char jsontemp[] =
            "{\"mip\":[10,0,0,0],\"sip\":[20,0,0,0],\"clname\":\"AERCM1\",\"ctrlt\":\"relayControl\",\"stst\":\"relayStatus\",\"clmt\":\"homeClimate\",\"mac\":[222,237,186,254,254,237]}";
```

and printed its content in the loop:
```
            memset(print_buffer, 0, sizeof(print_buffer));
            sprintf(print_buffer, "[%d]: %X\n",
                k, char_buf[k]);

            Serial.print(print_buffer);
```

The content of char buffer and JSON parsing are Ok:
```
[0]: 7B
[1]: 22
[2]: 6D
[3]: 69
[4]: 70
[5]: 22
[6]: 3A
[7]: 5B
[8]: 31
[9]: 30
[10]: 2C
[11]: 30
[12]: 2C
[13]: 30
[14]: 2C
[15]: 30
[16]: 5D
[17]: 2C
[18]: 22
[19]: 73
[20]: 69
...
```

Then I have replaced readString() to readBytes() as following:
```
        StaticJsonBuffer<200> json_buffer;
        get_config = false;
        memset(&module_config, 0, sizeof(module_config));
        char char_buf[190];
        byte k;
        do
        {
            k = Serial.available();
        }
        while(k == 0);
     
        Serial.readBytes(char_buf, k);
```

char buffer and JSON parser are not Ok:
```
[0]: 7B
[1]: 0
[2]: 0
[3]: 0
[4]: 0
[5]: 0
[6]: 0
[7]: 0
[8]: 0
[9]: 0
[10]: 0
[11]: 0
[12]: 0
[13]: 0
[14]: 0
[15]: 0
[16]: 0
[17]: 0
[18]: 0
[19]: 0
[20]: 0
[21]: 0
[22]: 0
...
[23]: 0
[24]: 0
[25]: 0
[26]: 0
[27]: 0
[28]: 0
[29]: 0
[30]: 0
[31]: 0
[32]: 0
[33]: 0
[34]: 0
[35]: 0
[36]: 0
[37]: 0
[38]: 0
[39]: 6D
[40]: 69
[41]: 70
[42]: 0
[43]: 31
[44]: 30
[45]: 0
[46]: 30
[47]: 0
[48]: 30
[49]: 0
[50]: 30
[51]: 0
[52]: 73
...
```

bblanchon, 2016-10-28:
It seems that the serial input is not coming at the right speed.
Did you check parity, baud rate... ?

Anyway, this issue is not related to ArduinoJson, there is really nothing I can do for you 😕

vitasam, 2016-10-28:
Yes, for sure it is not an ArduinoJson problem :)   So good to close this issue

closed 2016-10-28 by vitasam

vitasam, 2016-10-28:
Got this code working with long serial string and JSON parser:
```
        StaticJsonBuffer<SERIAL_BUF_SIZE> json_buffer;
        char char_buf[SERIAL_BUF_SIZE] = {0};
        bool in_loop = true;
        char received;

        // Clean serial input buffer
        while(Serial.available()) 
            Serial.read();
        
        k = 0;
        while(in_loop)
        {
            if (Serial.available())
            {
                if(k < SERIAL_BUF_SIZE)
                {
                    received = Serial.read();
                    if(received == '\n')                // Completed receiving
                        in_loop = false;
                    else
                        char_buf[k] = received;        
                    
                    k++;
                }
                else
                    in_loop = false;
            }
        }
             
        JsonObject& root = json_buffer.parseObject(char_buf);
```

bblanchon, 2016-10-29:
Good!
However, I see that there is a confusion between the size of the input and the size of the `JsonBuffer`.
The size of the `JsonBuffer` is what is needed to store the description of the JSON data.
You can use the [JsonBuffer size calculator](https://rawgit.com/bblanchon/ArduinoJson/master/scripts/buffer-size-calculator.html) to help you determine the required size.

vitasam, 2016-10-29:
Looks like 200 is enough for my string?

`{"mip":[10,20,30,40],"sip":[50,60,70,80],"clname":"AERCM1","ctrlt":"relayControl","stst":"relayStatus","clmt":"homeClimate","mac":[222,237,186,254,254,237]}`

Result is:

Platform	Size (in bytes)
AVR 8-bit	198

bblanchon, 2016-10-30:
The JSON string contains 157 characters, so `sizeof(char_buf)` should be at least 158.
Once again: 198, is the size of the `StaticJsonBuffer`.

vitasam, 2016-10-30:
Yes, that's clear now. I just made JSON and char buffers big enough, for my configuration string

bblanchon, 2016-10-30:
Good!
If you find that the documentation is misleading, please edit it 👍