bblanchon/ArduinoJson #378: Using arduinoJson outside arduino

zizyx, 2016-10-27:
Hello,

I'm using Json on arduino and it works like a charm, now I want to implement some unit tests around this but i cant seem to get them working on cmake and gtest, They should work without arduino libraries which is not the case. When i add the ArduinoJson directory to my cmake file it says it cant find the WString.h so i included that one then i asked for another file so i included that one too but after a while i got stuck on some avr/interrupt.h which i cant include for some reason. How do i turn off ArduinoJson so it wont use the arduino libary but instead just works for cpp?

bblanchon, 2016-10-27:
Hi,

ArduinoJson assumes it is built for Arduino as soon as `ARDUINO` is defined.

In your case, I think you just want to disable the support for `String`, so you should define `ARDUINOJSON_USE_ARDUINO_STRING` to `0`.

See: [ArduinoJson/Configuration.hpp](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Configuration.hpp).

zizyx, 2016-10-27:
Thank you very much, i was stuck on this for couple hours and im so releaved!

closed 2016-10-27 by zizyx

bblanchon, 2016-10-27:
You're welcome @zizyx, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

zizyx, 2016-10-28:
Have you ever tried to unit test your library? I get a segmentation error around 

```
0x00000000004498c0 in ArduinoJson::Internals::JsonParser::parseString (this=0x7fffffffd6e0)
    at /home/zizyx/Desktop/git-sensorNetwerk/sensor-network/gtest/lib/ArduinoJson/include/ArduinoJson/Internals/JsonParser.ipp:158
158	      *writePtr++ = c;
```

bblanchon, 2016-10-28:
There are currently 552 tests.
The continuous integration run them on 32 different compilers and platforms.
The code coverage is currently 99.843%
So yes, I ever tried to unit test my library.

But that doesn't mean there it's bug-free unfortunately.

Can I see your test's code?

zizyx, 2016-10-28:
Im sorry it wasn't my intention to sound mad ;) I was just curious

**Google test**
```
TEST(JSON, getInformation_and_unpack_it){
	JSON* json = new JSON();
	char* result = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
	json->unpackJSON(result);
}
```

**JSON.cpp**
```
void JSON::unpackJSON(char * entry){
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(entry);

    if(!root.success()){
        serial->println("parseObject() failed!");
        return;
    }

    const char* sensor    = root["sensor"];
    long        time      = root["time"];
    double      latitude  = root["data"][0];
    double      longitude = root["data"][1];

    serial->println(sensor);
    serial->println(time);
    serial->println(latitude);
    serial->println(longitude);
}
```

**Error from Gtest**
```
[----------] 2 tests from JSON
[ RUN      ] JSON.setInformation_and_pack_it
[       OK ] JSON.setInformation_and_pack_it (0 ms)
[ RUN      ] JSON.getInformation_and_unpack_it
Segmentation fault (core dumped)
```

The error in the previous comment is from the gdb debugger

bblanchon, 2016-10-28:
The error is here:

```c++
char* result = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
```

You should have the following warning on GCC:

```
warning: deprecated conversion from string constant to 'char*'
```

or on Clang:

```
warning: ISO C++11 does not allow conversion from string literal to 'char *' [-Wwritable-strings]
```

The solution is to declare an array instead:

```c++
char result[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
```

See:

* [Avoiding pitfalls: Make sure the string isn't read-only](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#7-make-sure-the-string-isnt-read-only)

zizyx, 2016-10-28:
Problem is that this solution still gives me the segmentation fault

bblanchon, 2016-10-28:
It's not a problem of type, it's what the pointer points to.

It needs to be writeable because the parser replaces some parts of the input string (that's how ArduinoJson works).
The string literal resides in a read-only data segment, that's why you have a segmentation fault.
By replacing with `char result[]` you make a copy of the string on the stack, which is writeable.

Are you sure that you still have a segfault after applying this change? This is very surprising.

bblanchon, 2016-10-28:
PS: [I added an entry in the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-do-i-have-a-segmentation-fault)

zizyx, 2016-10-28:
```
TEST(JSON, getInformation_and_unpack_it){
	JSON* json = new JSON();
	char result[54];
	// char result[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
	json->unpackJSON(result);
}
```

if I run this test it all runs fine except that i get the message that the parseObject() failed, which is obvious ofcourse. So I really think I'm put in the wrong data or I put my data into the wrong type.

bblanchon, 2016-10-28:
`JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(3)` is 192 according to the [calculator](https://rawgit.com/bblanchon/ArduinoJson/master/scripts/buffer-size-calculator.html), so 200 should be just enough.
However, this should not cause a segmentation fault, just a failing call to `parseObject()`.

zizyx, 2016-10-28:
well, whenever I found the fix ill let it know!

bblanchon, 2016-11-06:
⚠️ After upgrading to [ArduinoJson 5.7.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.7.0), you'll have to change:

```c++
#define ARDUINOJSON_USE_ARDUINO_STRING 0
```

into

```c++
#define ARDUINOJSON_ENABLE_STD_STRING 1
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 0
```

Sorry for the inconvenience.