bblanchon/ArduinoJson #379: memory overflow

marlonrettore, 2016-10-29:
Hello Friend,

I would like to ask you for help, I am doing a data transfer between two Arduino using NRF24L01 +, the program is working right, I get the data, I store in a JSON and send via serial to a third arduino. After making sending via serial, I'm cleaning using the RemoveAt () command to clear the fields created using createNestedArray, but after a while it stops filling the JSON, I believe that is bursting the buffer. Would like to help me, I do not know how to clean the JSON buffer.

bblanchon, 2016-10-29:
Hello,

A `JsonBuffer` should be used only once and then should be thrown away completely.
You don't need to clean it, you just declare it in a scope.
This is the way ArduinoJson is designed.

Please have a look at:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-shouldnt-i-use-a-global-jsonbuffer)
* [FAQ: How to reuse a JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-reuse-a-jsonbuffer)
* [FAQ: What's the best way to use the library?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)
* [Avoiding pitfalls: Don't reuse the same JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#4-dont-reuse-the-same-jsonbuffer)

If you still haven't found the relevant information, please paste a short code sample here.

marlonrettore, 2016-10-29:
Thanks for the answer,
I will do some tests if if does not work, I can send you the code so you can help me?

bblanchon, 2016-10-29:
Sure, I'll be glad to help.
But please keep the sample short and focused.

marlonrettore, 2016-10-29:
Goodnight friend,

just to give you a return for your test you showed me.
Gave quite sure, is now working very well and is not catching. Thanks for your help

bblanchon, 2016-10-30:
You're welcome @marlonrettore, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-10-30 by bblanchon