bblanchon/ArduinoJson #380: How to create Nested Object in Nested Array?

tarzan115, 2016-11-04:
Hi, Thank you for your Library.
I got a problem when create a JSON

here is what I want:
```
{
  "METHOD": "GET",  
  "ID": "123",
  "BTN": [
    {
      "ID":"1",
      "STT": "0"
    },
    {
    	"ID":"2",
       "STT": "1"
    },
    {
    	"ID":"3",
       "STT": "0"
    }
  ],
  "IP": "192.168.1.7",
  "VER": "Version",
  "PW": "Power",
  "TEMP": "Temperature"
}
```
and here is my code:
```
void sendJson(const ClientDATA& data, String json = "")
{
	StaticJsonBuffer<CLIENT_DATA_JSON_SIZE> jsonBuffer;
	JsonObject& root  = jsonBuffer.createObject();
	JsonArray& Button = root.createNestedArray("BTN");

	root["METHOD"]  = data.Method;
	root["ID"]		   = data.ID;
	root["IP"]		   = data.IP;
	root["VER"]	   = data.Version;
	root["PW"]	   = data.Power;
	root["TEMP"]	   = data.Temperature;

	JsonObject& Button1 = Button.createNestedObject();
	Button1["ID"] = data.ButtonID[0];
	Button1["STT"] = data.Status[0];

	JsonObject& Button2 = Button.createNestedObject();
	Button2["ID"]   = data.ButtonID[1];
	Button2["STT"] = data.Status[1];

	JsonObject& Button3 = Button.createNestedObject();
	Button3["ID"]   = data.ButtonID[2];
	Button3["STT"] = data.Status[2];
	
        // for(int i=0; i<3; i++){
	// 	JsonObject& ButtonID = Button.createNestedObject();
	// 	ButtonID["ID"] = data.ButtonID[i];
	// 	ButtonID["STT"] = data.Status[i];
	// }

	// // Button.add(ButtonID);

	// ButtonID["ID"] = data.ButtonID[1];
	// ButtonID["STT"] = data.Status[1];
	// // Button.add(ButtonID);

	// ButtonID["ID"] = data.ButtonID[2];
	// ButtonID["STT"] = data.Status[2];
	// Button.add(ButtonID);

	// root.printTo(json);
	root.prettyPrintTo(Serial);
}
```

I also try loop:
```
for(int i=0; i<3; i++){
	 	JsonObject& ButtonID = Button.createNestedObject();
	 	ButtonID["ID"] = data.ButtonID[i];
	 	ButtonID["STT"] = data.Status[i];
	 }
```

But I always got 2 elements of __*"BTN"*__ like:
```
{
  "METHOD": "GET",  
  "ID": "123",
  "BTN": [
    {
      "ID":"1",
      "STT": "0"
    },
    {
    	"ID":"2",
       "STT": "1"
    }
  ],
  "IP": "192.168.1.7",
  "VER": "Version",
  "PW": "Power",
  "TEMP": "Temperature"
}
```
please explain how to create that JSON.
thank you so much.

bblanchon, 2016-11-04:
Hi Doanh,

The `StaticJsonBuffer` is probably too small.

See:

* [FAQ: Why some parts are missing?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-some-parts-are-missing)
* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)
* [JsonBuffer size calculator](https://rawgit.com/bblanchon/ArduinoJson/master/scripts/buffer-size-calculator.html)

tarzan115, 2016-11-05:
thank you so so much ❤️ 
I made a mistake when use JSON_OBJECT_SIZE.
I fixed. ^_^
thank you.

closed 2016-11-05 by tarzan115