bblanchon/ArduinoJson #381: Failed compile using 5.7.0

sticilface, 2016-11-12:
I'm not entirely sure what is going on here, or why, but this line causes my compilation to fail.  whereas the same line in an example ino does not.

```
set.STA.MAC[i] = STAjson[string_MAC][i]; 
```

set.STA.MAC is a 6 byte uin8_t array.  but using a temp uint8_t or anything else does not work either. 

STAjson is defined as 
```
        JsonObject & STAjson = root[string_STA].as<JsonObject>();
```

the string_ are static const char * to simple strings...  

the error is below

```
In file included from .piolibdeps/ArduinoJson_ID64/include/ArduinoJson.hpp:12:0,
from .piolibdeps/ArduinoJson_ID64/include/ArduinoJson.h:8,
from .piolibdeps/ArduinoJson_ID64/ArduinoJson.h:8,
from lib/ESPManager/src/ESPmanager.h:48,
from lib/ESPManager/src/ESPmanager.cpp:2:
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type* ArduinoJson::JsonObject::getNodeAtImpl(TStringRef) const [with TStringRef = const unsigned char&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type = ArduinoJson::Internals::ListNode<ArduinoJson::JsonPair>]':
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:141:68:   required from 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type* ArduinoJson::JsonObject::getNodeAt(const TString&) const [with TString = unsigned char; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type = ArduinoJson::Internals::ListNode<ArduinoJson::JsonPair>]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:93:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObject::get(const TString&) const [with TValue = unsigned char; TString = const unsigned char&; typename ArduinoJson::Internals::JsonVariantAs<T>::type = unsigned char]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Internals/../JsonObjectSubscript.hpp:52:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = unsigned char; TString = unsigned char; typename ArduinoJson::Internals::JsonVariantAs<T>::type = unsigned char]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Internals/../JsonVariantBase.hpp:60:35:   required from 'const typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = unsigned char; TImpl = ArduinoJson::JsonObjectSubscript<unsigned char>; typename ArduinoJson::Internals::JsonVariantAs<T>::type = unsigned char]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Internals/../JsonVariantBase.hpp:55:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = unsigned char; TImpl = ArduinoJson::JsonObjectSubscript<unsigned char>]'
lib/ESPManager/src/ESPmanager.cpp:3028:32:   required from here
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:147:76: error: 'equals' is not a member of 'ArduinoJson::Internals::StringFuncs<const unsigned char&>'
if (Internals::StringFuncs<TStringRef>::equals(key, node->content.key))
^
*** [.pioenvs/bedroom/lib/ESPManager/ESPmanager.o] Error 1
```

it happens in platformio and arduino, and only for the latest 5.7.0 version. 

if i change the code to 
```

            JsonArray & mac = STAjson[string_MAC];

            for (uint8_t i = 0; i < 6; i++) {
                set.STA.MAC[i] = mac[i]; 
            }
```

then it works.  

bit of a loss really

sticilface, 2016-11-12:
OK the issue seems to reside in the use of F() macro... 

although not used there, i have them spotted around, and even though it is meant to support

JsonArray & mac = STAjson[F("abc")];

this does not work.

```
In file included from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.hpp:12:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.h:8,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/arduino_modified_sketch_423468/JsonGeneratorExample.ino:8:
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type* ArduinoJson::JsonObject::getNodeAtImpl(TStringRef) const [with TStringRef = const __FlashStringHelper*; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type = ArduinoJson::Internals::ListNode<ArduinoJson::JsonPair>]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:141:68:   required from 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type* ArduinoJson::JsonObject::getNodeAt(const TString&) const [with TString = const __FlashStringHelper*; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type = ArduinoJson::Internals::ListNode<ArduinoJson::JsonPair>]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:93:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObject::get(const TString&) const [with TValue = ArduinoJson::JsonArray&; TString = const __FlashStringHelper*; typename ArduinoJson::Internals::JsonVariantAs<T>::type = ArduinoJson::JsonArray&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/../JsonObjectSubscript.hpp:52:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = ArduinoJson::JsonArray&; TString = const __FlashStringHelper*; typename ArduinoJson::Internals::JsonVariantAs<T>::type = ArduinoJson::JsonArray&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/../JsonVariantBase.hpp:60:35:   required from 'const typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = ArduinoJson::JsonArray&; TImpl = ArduinoJson::JsonObjectSubscript<const __FlashStringHelper*>; typename ArduinoJson::Internals::JsonVariantAs<T>::type = ArduinoJson::JsonArray&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/../JsonVariantBase.hpp:38:14:   required from 'ArduinoJson::JsonArray& ArduinoJson::JsonVariantBase<TImpl>::asArray() const [with TImpl = ArduinoJson::JsonObjectSubscript<const __FlashStringHelper*>]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/../JsonArraySubscript.hpp:73:18:   required from 'const ArduinoJson::JsonArraySubscript ArduinoJson::JsonVariantBase<TImpl>::operator[](int) const [with TImpl = ArduinoJson::JsonObjectSubscript<const __FlashStringHelper*>]'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/arduino_modified_sketch_423468/JsonGeneratorExample.ino:55:30:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:147:76: error: 'equals' is not a member of 'ArduinoJson::Internals::StringFuncs<const __FlashStringHelper*>'
       if (Internals::StringFuncs<TStringRef>::equals(key, node->content.key))
```

bblanchon, 2016-11-13:
Hi @sticilface,

Thank you very much for reporting this issue.

You're right about the second error, it is related to the `F()` macro.
PROGMEM is not supported in ArduinoJson 5.7.0, it's issue #76.
However, it should be available in 5.7.1, see [branch issue76](https://github.com/bblanchon/ArduinoJson/tree/issue76).

The error you had in the first message is different.
It is a new bug, I'm working on it.

sticilface, 2016-11-13:
no problem.  2 issues explains why i was having a bit of a nightmare tracking it down.  this was also only occurring when i used a beta feature of platoformio with remote compilation. as it would automatically download the latest lib eery time, and not use the cached one which my local platformio was doing.   got there in the end though!

bblanchon, 2016-11-13:
Sorry about that.
Once again, thanks for taking the time to report the bug, you've been very helpful with the community.

I'll merge the two fixes, as soon as the builds pass.

sticilface, 2016-11-13:
no problem!  totally happy to help!  its a great lib:) and i use it in everything!

bblanchon, 2016-11-13:
Fixed in ArduinoJson 5.7.1.

closed 2016-11-13 by bblanchon

sticilface, 2016-11-13:
unfortunately...  it is not fixed :(

Are you able to compile using Arduino and my example from here 
https://github.com/sticilface/ESPmanager

here is the error

```
In file included from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.hpp:12:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.h:8,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.h:48,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp:2:
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type* ArduinoJson::JsonObject::getNodeAtImpl(TStringRef) const [with TStringRef = const __FlashStringHelper*; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type = ArduinoJson::Internals::ListNode<ArduinoJson::JsonPair>]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:155:52:   required from 'bool ArduinoJson::JsonObject::setNodeAt(TStringRef, TValueRef) [with TStringRef = const __FlashStringHelper*; TValueRef = const bool&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:78:19:   required from 'bool ArduinoJson::JsonObject::set(const TString&, const TValue&) [with TValue = bool; TString = const __FlashStringHelper*]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/../JsonObjectSubscript.hpp:42:5:   required from 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = bool; TString = const __FlashStringHelper*]'
/Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp:1416:35:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:147:76: error: 'equals' is not a member of 'ArduinoJson::Internals::StringFuncs<const __FlashStringHelper*>'
       if (Internals::StringFuncs<TStringRef>::equals(key, node->content.key))
                                                                            ^
In file included from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonArray.hpp:15:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.hpp:11,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.h:8,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.h:48,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp:2:
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/ValueSetter.hpp: In instantiation of 'static bool ArduinoJson::Internals::ValueSetter<TSource, Enable>::set(ArduinoJson::JsonBuffer*, TDestination&, const TSource&) [with TDestination = const char*; TSource = const __FlashStringHelper*; Enable = void]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:161:42:   required from 'bool ArduinoJson::JsonObject::setNodeAt(TStringRef, TValueRef) [with TStringRef = const __FlashStringHelper*; TValueRef = const bool&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:78:19:   required from 'bool ArduinoJson::JsonObject::set(const TString&, const TValue&) [with TValue = bool; TString = const __FlashStringHelper*]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/../JsonObjectSubscript.hpp:42:5:   required from 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = bool; TString = const __FlashStringHelper*]'
/Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp:1416:35:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Internals/ValueSetter.hpp:23:17: error: cannot convert 'const __FlashStringHelper* const' to 'const char*' in assignment
     destination = source;
```

bblanchon, 2016-11-14:
My mistake. I assumed that PROGMEM was only available on AVR.

You'll need to add

    #define ARDUINOJSON_ENABLE_PROGMEM 1

at the top of the file, until I fix this in the next release.

sticilface, 2016-11-14:
ah that does indeed fix the issue.  thank you