bblanchon/ArduinoJson #382: error: invalid conversion from 'const char*' to 'int' [-fpermissive]

dmsherazi, 2016-11-15:
I am getting this error  `error: invalid conversion from 'const char*' to 'int' [-fpermissive]` on 

`const char *user = root["p"]["u"];`

root is passed in to the function in this way  `void addAdmin(JsonObject &root) {`

Here is the code for the function 

````
void addAdmin(JsonObject &root) {// un-subscribe from Default topic
    // At the moment is not required as we will disconnect
    // client.unsubscribe(defSubScribeTopic);

    const char *regTopic = root[PARAMS]["r"];
    setGeneratedTopic(regTopic, sizeOfGenTopic);

    //Save admin number to EEPROM with null as terminating character
    const char *user = root[PARAMS][U_UNAME];
    // make sure we don't cross into another user memory
    // write all the bytes there is no issue until we have enough space
    eepu(user, users[0], maxChar4Number);
    /*eeprom_update_byte(users[strlen(user)], 0);
    eeprom_update_byte(&callPermission[0], 1);*/
    writepgmstatus(PROGRAMMED);
    setAdminAckSent(0);
    //???  All stuff below is useless if we need to restart
    // Get the new Subscribe and Publish topics
    //getGeneratedTopic(subScribeTopic, false);
    //client.unsubscribe(deaultt)
    //s800.switchModem();
    // reattemptConnection(CONNECT_MQTTT);
    programmingSegmentOn;
    wdt_reset();
}
````

and here is the complete error log
````
src\main.cpp:132:44: error: invalid conversion from 'const char*' to 'int' [-fpermissive]
const char *user = root["p"]["u"];
^
In file included from lib\ArduinoJson/ArduinoJson/Internals/JsonSerializer.ipp:11:0,
from lib\ArduinoJson/ArduinoJson.hpp:16,
from lib\ArduinoJson/ArduinoJson.h:8,
from src\main.cpp:5:
lib\ArduinoJson/ArduinoJson/Internals/../JsonArraySubscript.hpp:71:33: error:   initializing argument 1 of 'const ArduinoJson::JsonArraySubscript ArduinoJson::JsonVariantBase<TImpl>::operator[](int) const [with TImpl = ArduinoJson::JsonObjectSubscript<char []>]' [-fpermissive]
inline const JsonArraySubscript JsonVariantBase<TImplem>::operator[](
^
````

bblanchon, 2016-11-15:
Do you have this error with ArduinoJson 5.6 too?

dmsherazi, 2016-11-15:
I will check it now and see

dmsherazi, 2016-11-15:
Changed to `ArduinoJson-5.6.7` and teh problem is solved.

bblanchon, 2016-11-16:
@dmsherazi I do not manage to reproduce the bug.

Can you provide a complete (but short) program reproducing the issue?
What compiler are you using?
What is the target platform?
What version of ArduinoJson were you using 5.7.0 or 5.7.1?

bblanchon, 2016-11-24:
@dmsherazi I need your help to reproduce this bug.
Can you please answer the questions of my previous message?

dmsherazi, 2016-11-25:
I will do it tomorrow.

Best Regards,

              <http://www.dostmuhammad.com/>

Dost Muhammad Shah / Engr
info@dostmuhammad.com
engr.dost.mhd@gmail.com

+968 9777-5603
www.dostmuhammad.com

[image: Twitter] <http://twitter.com/dmsherazi> [image: Facebook]
<https://www.facebook.com/Dost.Muhammad.Shah> [image: Google +]
<https://plus.google.com/u/0/+DostMuhammadShah/posts> [image: LinkedIn]
<https://www.linkedin.com/in/dmsherazi> [image: Flickr]
<https://www.flickr.com/photos/54804393@N02/> [image: Github]
<https://github.com/dmsherazi> [image: Stack Overflow]
<http://stackoverflow.com/a/12335916/1155507> [image: Quora]
<https://www.quora.com/profile/Dost-Muhammad-Sherazi>

bblanchon, 2016-12-01:
Any progress?

bblanchon, 2016-12-09:
I'm closing this issue since there has been no new information for 24 days.
Please reopen if needed.

Gatux, 2016-12-19:
Hi,
I have the same problem with versions 5.7.1, 5.7.2 and 5.7.3
Seems to be working with version 5.7.0

Tested wih esp8266 12-F with PROGMEM :
`json[JSON_DATA][JSON_HOSTNAME]`

bblanchon, 2016-12-20:
Hi @Gatux,

I'm quite curious about what's hidden behind `PROGMEM_DATA` and `PROGMEM_HOSTNAME`...

Did you try the [ProgmemExample](https://github.com/bblanchon/ArduinoJson/blob/master/examples/ProgmemExample/ProgmemExample.ino)?
Can you please provide a minimalistic program that reproduces the bug?
Can you please copy and paste the compiler's output?

Thank you very much.

Gatux, 2016-12-20:
So, after new tests today, I think I did something wrong last nigth in order to get the compilation problem.

Working now with v5.7.3, and using progmem properly, I am able to make it all work:
```
const char key[] PROGMEM = "key";
const char subkey[] PROGMEM = "subkey";
Serial.println(root[FPSTR(key)].asString());
Serial.println(root[FPSTR(key)][FPSTR(subkey)].asString());
```

Thanks for the help, and sorry for inconvenience.

bblanchon, 2016-12-20:
The problem is that `PROGMEM` is not a type but an attribute; therefore I'm not sure the right overload is selected.

You should use the `F()` macros instead, because it casts to a `const __FlashStringHelper *`.

```
Serial.println(root[F("key")].asString());
Serial.println(root[F("key")][F("subkey")].asString());
```

closed 2016-12-20 by bblanchon