bblanchon/ArduinoJson #383: git head, crash on using F() on ESP8266

sticilface, 2016-11-16:
get the following stack trace on any F() use

```
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4023349c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40264fed, depc=0x00000000

Exception (3):
epc1=0x4023349c epc2=0x00000000 epc3=0x00000000 excvaddr=0x40264fed depc=0x00000000

ctx: sys 
sp: 3ffff690 end: 3fffffb0 offset: 01a0
```

with it decoded

```
Decoding 78 results
0x4010020c: _umm_free at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x40219233: bool ArduinoJson::JsonObject::setNodeAt__FlashStringHelper const*, bool const&>(__FlashStringHelper const*, bool const&) at /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp line 893
0x4021c04d: ESPmanager::_HandleDataRequest(AsyncWebServerRequest*) at /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp line 893
0x401004ba: _umm_malloc at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1462
0x4022f84c: String::move(String&) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x402327d7: operator()  at /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp line 893
:  (inlined by) __call  at /Users/amelvin/git/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 1296
:  (inlined by) operator()  at /Users/amelvin/git/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 1355
:  (inlined by) std::_Function_handler    (ESPmanager*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*) at /Users/amelvin/git/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 2071
0x402101a1: std::function ::operator()(AsyncWebServerRequest*) const at /Users/amelvin/git/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 2439
:  (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebHandlerImpl.h line 89
0x402327d0: std::_Function_base::_Base_manager    (ESPmanager*, std::_Placeholder1>)> >::_M_get_pointer(std::_Any_data const&) at /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp line 893
:  (inlined by) std::_Function_handler    (ESPmanager*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*) at /Users/amelvin/git/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 2071
0x4010053d: _umm_realloc at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1709
0x4022f75b: String::reserve(unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4010020c: _umm_free at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x4010068c: free at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1733
0x402327f1: std::_Function_handler    (ESPmanager*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*) at /Users/amelvin/Documents/Arduino/libraries/ESPmanager/src/ESPmanager.cpp line 893
0x402101a1: std::function ::operator()(AsyncWebServerRequest*) const at /Users/amelvin/git/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 2439
:  (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebHandlerImpl.h line 89
0x4020e724: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022fa8d: String::concat(char const*, unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4020e3d6: AsyncWebServerRequest::_parseLine() at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
0x4022faf0: String::concat(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 720
0x4020e5c8: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/amelvin/Documents/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 602
[... cut ...]
0x40258c6d: ip_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip.c line 559
0x4023d65e: ieee80211_hostap_attach at ?? line ?
0x4023db52: hostap_input at ?? line ?
0x40258065: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c line 1379
0x4010594e: wdt_feed at ?? line ?
0x4024cea8: ets_timer_handler_isr at ?? line ?
0x4024cece: ets_timer_handler_isr at ?? line ?
0x4024cb2b: ets_snprintf at ?? line ?
```

sorry this is turning into a pain.  

the same sketch has no crash using 5.6.7

bblanchon, 2016-11-17:
I'll try that on an ESP board tonight.
How did you get the stack trace?

sticilface, 2016-11-17:
I use this tool

https://github.com/me-no-dev/EspExceptionDecoder

very useful.

bblanchon, 2016-11-18:
I spent one hour trying to upload a sketch to my SparkFun ESP8266 Thing, but I failed.
I ordered an Adafruit Feather Huzzah, but I will only receive it next week.

I tried to understand what's going on by looking at the stack trace and the source of the ESP8266 core, but I don't get any clue.

Can you post the smallest possible program to reproduce the issue?

aacton, 2016-11-19:
@bblanchon You mentioned having trouble uploading your sketch to an ESP8266. Are you using a good power supply or just a USB->serial converter like an FTDI? My experience has been that many (most?) stability issues with the ESP8266 are related to a lack of power. If you aren't already, I recommend using a good external power supply like the YwRobot MB102. It made a huge difference for me.

bblanchon, 2016-11-23:
This is fixed in release 5.7.2 ([see the ridiculous commit](https://github.com/bblanchon/ArduinoJson/commit/2f7232859e2eaebca561b08c3ff4b2f2335493ea))

@aacton Thank for your suggestion. After your message I tried again with a different power cable but it didn't work; maybe it's my FTDI that is fried.

closed 2016-11-23 by bblanchon

sticilface, 2016-11-23:
Hi,

Sorry for my absence over the last few days.  my job was taking most of my time... 13hr days over the weekend so i just did not get a chance!  That is  ridiculous commit... i will test it now.. on my shiny new mac!😊