bblanchon/ArduinoJson #384: How to delect a Object

waayst, 2016-11-17:
with StaticJsonBuffer we can use parseObject but After a few times parse,it will full and won't work,so how to
delect a useless object.

bblanchon, 2016-11-17:
The problem is that you're reusing the same JsonBuffer several times.
Please, have a look at the FAQ.

bblanchon, 2016-11-18:
See:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-shouldnt-i-use-a-global-jsonbuffer)
* [FAQ: How to reuse a JsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-reuse-a-jsonbuffer)
* [FAQ: What's the best way to use the library?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)

closed 2016-11-24 by bblanchon