bblanchon/ArduinoJson #385: Issue when trying to use custom Print implementation

cosmikwolf, 2016-11-18:
I am trying to use a custom Print implementation, but I am running into an error: 

`lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:55:3: error: 'has_append' is not a member of 'ArduinoJson::Internals::StringFuncs<ChunkPrint (Flas
hMemory::*)(SerialFlashFile)>'`

I have implemented the following functions into my implementation:
`virtual size_t write(uint8_t c)
virtual size_t write(const uint8_t *buffer, size_t size)`

I feel like I must be missing something though.. Can you help point me in the right direction?

bblanchon, 2016-11-18:
I tried the [ChunkPrint class from the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks#chunked-output) and it works like a charm.

I bet that you forgot to inherit from `Print`, because I manage to get the same error when I change:

    class ChunkPrint : public Print {

to:

    class ChunkPrint {

cosmikwolf, 2016-11-18:
I do have the Print inherit properly in my header file, but I think you are right in that for some reason the compiler doesn't it is a Print output. It thinks its a String... 

Here is my header file, I will continue to experiment with things and let you know if I come up with a solution.
`#include <Arduino.h>
#include <SerialFlash.h>
#include "Print.h"

#ifndef _ChunkPrint_h_
#define _ChunkPrint_h_

class ChunkPrint : public Print {
 public:
  ChunkPrint(SerialFlashFile * file)
      : _file(file), _current_byte(0) {}

      virtual size_t write(uint8_t c) {
        if(_current_byte == 0){
          _file->erase();
          _file->seek(0);
        }
        sprintf(buf, "%c", c);
        _file->write(buf, 1);
        _current_byte++;
        return 0;
      }

      virtual size_t write(const uint8_t *buffer, size_t size){
        _file->erase();
        _file->seek(0);
        _file->write(buffer, size);
        _file->write(NULL, 1);
        return 0;
      };

 private:
  char buf[1];
  uint16_t _current_byte;
  SerialFlashFile * _file;
};

#endif
`

cosmikwolf, 2016-11-18:
Ah, for some reason when I split up the header into a .h and .cpp file, it compiles fine! My guess is that the constructor is not built properly... 

To explain a bit for anyone else who may read this in the future, this code is designed to create a Print interface for SerialFlash, to allow me to Print directly to an SPI flash chip without having to dump everything into a char buffer in order to save memory.

Here is the code that compiles OK:

ChunkPrint.h:
```
#include <Arduino.h>
#include <SerialFlash.h>
#include <Print.h>

#ifndef _ChunkPrint_h_
#define _ChunkPrint_h_

class ChunkPrint : public Print {
  public:
    ChunkPrint(SerialFlashFile * file);

    virtual size_t write(uint8_t c);

    virtual size_t write(const uint8_t *buffer, size_t size);
  private:
    char buf[1];
    uint16_t _current_byte;
    SerialFlashFile * _file;
};

#endif
```

ChunkPrint.cpp
```#include "ChunkPrint.h"

ChunkPrint::ChunkPrint(SerialFlashFile * file){
  _file = file;
  _current_byte = 0;
}

size_t ChunkPrint::write(uint8_t c) {
  if(_current_byte == 0){
    _file->erase();
    _file->seek(0);
  }
  sprintf(buf, "%c", c);
  _file->write(buf, 1);
  _current_byte++;
  return 0;
}

size_t ChunkPrint::write(const uint8_t *buffer, size_t size){
  _file->erase();
  _file->seek(0);
  _file->write(buffer, size);
  _file->write(NULL, 1);
  return 0;
};
```

bblanchon, 2016-11-18:
OK, that's very curious.
I don't understand why the compiler didn't choose the right overload of `printTo()` in the first place.

Next time this happens, please have a look at the lines before `error: 'has_append' is not a member...`, they usually tell why an overload has been rejected.

Here is an example after removing the reference to the source code:

```
error: no matching function for call to 'ArduinoJson::JsonObject::printTo(ChunkPrint&)'
note: candidates are:
note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(Print&) const [with T = ArduinoJson::JsonObject; size_t = unsigned int]
note:   no known conversion for argument 1 from 'ChunkPrint' to 'Print&'
note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(char*, size_t) const [with T = ArduinoJson::JsonObject; size_t = unsigned int]
note:   candidate expects 2 arguments, 1 provided
note: template<class TString> typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::Internals::StringFuncs<TString>::has_append, unsigned int>::type ArduinoJson::Internals::JsonPrintable<T>::printTo(TString&) const [with TString = TString; T = ArduinoJson::JsonObject]
note:   template argument deduction/substitution failed:
error: 'has_append' is not a member of 'ArduinoJson::Internals::StringFuncs<ChunkPrint>'
```

cosmikwolf, 2016-11-19:
I take back what I said about it compiling and working earlier. I think I messed something up with my files, and re-wrote everything. Now I get the same error again. 

I think the <unresolved overloaded function type> in the beginning is the key. For some reason the serialFlashPrint (I renamed chunkPrint) is not being recognized as Print&.

```
src/FlashMemory.cpp: In member function 'void FlashMemory::saveSequenceJSON(uint8_t, uint8_t)':
src/FlashMemory.cpp:98:36: error: no matching function for call to 'ArduinoJson::JsonObject::printTo(<unresolved overloaded function type>)'
root.printTo(serialFlashPrint);
^

src/FlashMemory.cpp:98:36: note: candidates are:
In file included from lib/ArduinoJson/include/ArduinoJson/Internals/../JsonVariant.hpp:13:0,
from lib/ArduinoJson/include/ArduinoJson/Internals/../JsonBuffer.hpp:14,
from lib/ArduinoJson/include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
from lib/ArduinoJson/include/ArduinoJson/DynamicJsonBuffer.hpp:10,
from lib/ArduinoJson/include/ArduinoJson.hpp:10,
from lib/ArduinoJson/include/ArduinoJson.h:8,
from lib/ArduinoJson/ArduinoJson.h:8,
from src/FlashMemory.h:5,
from src/FlashMemory.cpp:2:
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:34:10: note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(Print&) const [with T = Ardu
inoJson::JsonObject; size_t = unsigned int]
size_t printTo(Print &print) const {
^
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:34:10: note:   no known conversion for argument 1 from '<unresolved overloaded function type>' to 
'Print&'
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:48:10: note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(char*, size_t) const [with T
 = ArduinoJson::JsonObject; size_t = unsigned int]
size_t printTo(char *buffer, size_t bufferSize) const {
^
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:48:10: note:   candidate expects 2 arguments, 1 provided
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:55:3: note: template<class TString> typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::Intern
als::StringFuncs<TString>::has_append, unsigned int>::type ArduinoJson::Internals::JsonPrintable<T>::printTo(TString&) const [with TString = TString; T = ArduinoJson::JsonObje
ct]
printTo(TString &str) const {
^
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:55:3: note:   template argument deduction/substitution failed:
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp: In substitution of 'template<class TString> typename ArduinoJson::TypeTraits::EnableIf<ArduinoJso
n::Internals::StringFuncs<TString>::has_append, unsigned int>::type ArduinoJson::Internals::JsonPrintable<T>::printTo(TString&) const [with TString = TString; T = ArduinoJson:
:JsonObject] [with TString = SerialFlashPrint (FlashMemory::*)(SerialFlashFile*)]':
src/FlashMemory.cpp:98:36:   required from here
lib/ArduinoJson/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:55:3: error: 'has_append' is not a member of 'ArduinoJson::Internals::StringFuncs<SerialFlashPrint
 (FlashMemory::*)(SerialFlashFile*)>'
```

cosmikwolf, 2016-11-19:
I submitted a question about this to Stack Overflow: http://stackoverflow.com/questions/40688280/c-polymorphic-print-function-not-resolving-properly

closed 2016-11-30 by bblanchon