bblanchon/ArduinoJson #387: Edit a value on the fly?

FinlayDaG33k, 2016-11-20:
Hii there,

I'm working on a project for a friend of mine.
However, Iḿ running against a small issue.
I seem to not be able to edit a value of the json on the fly?

this is the code that I have when encoutering the issue: (link to gist because well... it's less of a mess...)
[Link to the Gist](https://gist.github.com/FinlayDaG33k/bd192b71141b4e96841050ed106fbff7)

As you can see, in the R1 function, I try to edit some values on the fly (for example `variables["R1"]["raw"]`).

However, I get an error thrown at my face:

    /home/finlay/Arduino/Cas/Sketch_4/Sketch_4.ino: In function 'void R1()':
    Sketch_4:16: error: no match for 'operator*' (operand types are 'ArduinoJson::TypeTraits::EnableIf<true, const ArduinoJson::JsonObjectSubscript<char [4]> >::type {aka const ArduinoJson::JsonObjectSubscript<char [4]>}' and 'const int')
     buffer= variables["R1"]["raw"] * Vin;
                                    ^
    Sketch_4:19: error: no match for 'operator*' (operand types are 'ArduinoJson::TypeTraits::EnableIf<true, const ArduinoJson::JsonObjectSubscript<char [15]> >::type {aka const ArduinoJson::JsonObjectSubscript<char [15]>}' and 'float')
     variables["R1"]["sensorValue"]= variables["R1"]["sensorResistor"] * buffer;
                                                                       ^
    exit status 1
    no match for 'operator*' (operand types are 'ArduinoJson::TypeTraits::EnableIf<true, const ArduinoJson::JsonObjectSubscript<char [4]> >::type {aka const ArduinoJson::JsonObjectSubscript<char [4]>}' and 'const int')

How can I edit the values of the JSON on the fly? (mainly whats inside the `variables` array)

bblanchon, 2016-11-20:
Hi Aroop,

The compiler complains because it doesn't know how to multiply a `JsonVariant` by an `int`.
Indeed, the type of this expression depends on what's in the `JsonVariant` at run-time and cannot be guessed at compile-time.
Solution? You need to tell the compiler what type you expect the `JsonVariant` to be:

```c++
buffer = variables["R1"]["raw"].as<int>() * Vin;
// or
buffer = (int)variables["R1"]["raw"] * Vin;
```

FinlayDaG33k, 2016-11-20:
Ah yes, it seems to compile now :) 
Thanks!

closed 2016-11-20 by FinlayDaG33k

bblanchon, 2016-11-20:
You're welcome Aroop, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: