bblanchon/ArduinoJson #388: Iterate trough nested array

FinlayDaG33k, 2016-11-21:
Hii there,

I'm trying to loop trough a nested array.
my array looks like [this](https://gist.github.com/FinlayDaG33k/d9acf68de84aff957a1b57bc08ab0db1).

when I use this code: (`variables` is the name of my parsed JSON, just like `root` is in the examples)

    for (JsonObject::iterator it=variables.begin(); it!=variables.end(); ++it){
        Serial.println(it->key);
        Serial.println(it->value.asString());
    }

I get this output in the serial monitor:

    Vin
    5
    trigger
    45000
    detrigger
    43000
    fingers

However, I want it to look like this:

    R1
    R2
    R3
    L1
    L2
    L3

How can I achieve this?

bblanchon, 2016-11-21:
Hi Aroop,

It's easy, just iterate on the `fingers` object instead of the root object:

```c++
JsonObject& fingers =  variables["finger"];
for (auto finger : fingers){
    Serial.println(finger.key);
}
```

FinlayDaG33k, 2016-11-21:
Sweet, that seems to work :) (gonna add it to the Wiki in a moment)

closed 2016-11-21 by FinlayDaG33k

MikeyMoMo, 2017-11-14:
I am really not getting this.  Is there an example of an ESP8266/Arduino generalized parser for a json packet that will print all values from all levels and get a count of how many keys on each level.  My head is spinning on this.  I cannot figure the syntax to get to the second level.