bblanchon/ArduinoJson #390: Comparing 2 JSON outputs?

FinlayDaG33k, 2016-11-25:
Hii there,

I'm working on my project again, and came across a small issue:
I want to ONLY send data when the old JSON output differs from the new/current one.

So it goes a bit like this:

1. Create the JSON and output in the program
2. check if the previously generated JSON differs from the new one.
2a. if previously generated JSON differs, send new JSON through the serial then update the old JSON with the current one
2b. if previously generated JSON is the same, do nothing until next loop

This is to make sure that somethink has actually changed in the JSON before sending it out (so save up some bandwidth and stuff).

Is this possible?

bblanchon, 2016-11-25:
Hi,

I suggest that you compute the hash of the JSON before sending it.
This way you just have to compare the hash of the previous message with the new one.

See: [Bag of Tricks: Compute hash of JSON output](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks#compute-hash-of-json-output)

FinlayDaG33k, 2016-11-25:
I thought so yea, but I don't know how the Arduino will handle it to do this a few hundred times a second

bblanchon, 2016-11-25:
Maybe you should compare the data before creating the `JsonObject`, that would save a lot of cycles.
You could also keep a `bool dirty` that is set to true each time data is modified.

FinlayDaG33k, 2016-11-25:
Yes, makes sense :) I'll try that out :) (will close the ticket if it works)

closed 2016-11-25 by FinlayDaG33k