bblanchon/ArduinoJson #391: Json parse fails when string is array containing one object?

joeybab3, 2016-11-26:
I end up with a string like this:
[{"id":"d073d5028b80","uuid":"24fcd5-238461903764","label":"Room","connected":false,"power":"on","color":{"hue":251.38200961318378,"saturation":0.009994659342336155,"kelvin":2500},"brightness":0.5699855039291981,"group":{"id":"72ca5c93acd491a7a757ed28483ffce8","name":"Room 1"},"location":{"id":"75addaabeb82666108e8e70c98ae7bec","name":"My Home"},"product":{"name":"Original 1000","identifier":"lifx_original_a21","company":"LIFX","capabilities":{"has_color":true,"has_variable_color_temp":true,"has_ir":false,"has_multizone":false}},"infrared":null,"last_seen":"2016-11-26T09:30:43.000+00:00","seconds_since_seen":100.602491339}]
However, root.success() fails. I can't do something like 
    String id = root[0]["id"];
because it throws errors.

bblanchon, 2016-11-27:
You need to call `parseArray()`, not `parseObject()`.

I opened a new issue about the crash.

See:

* [FAQ: Why parsing fails](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* Issue #392

joeybab3, 2016-11-27:
Sorry, I thought I had already looked through the issues, thanks for your help!

closed 2016-11-27 by joeybab3

bblanchon, 2016-11-28:
You're welcome Joey, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: