bblanchon/ArduinoJson #393: Quite high flash usage with STM32Duino

maichaell, 2016-11-29:
I'm using STM32Duino "framework" with Maple Mini- board and would like to use this library.
https://github.com/rogerclarkmelbourne/Arduino_STM32

The sketch is taken from an example. 

`#include "ArduinoJson.h"
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(9600);
  // initialize digital pin 13 as an output.
  pinMode(33, OUTPUT);
  
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(33, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(33, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second

  StaticJsonBuffer<200> jsonBuffer;
  
  JsonObject& root = jsonBuffer.createObject();
  root["sensor"] = "gps";
  root["time"] = 1351824120;
  
  JsonArray& data = root.createNestedArray("data");
  data.add(48.756080, 6);  // 6 is the number of decimals to print
  data.add(2.302038, 6);   // if not specified, 2 digits are printed
  
  //
  // Step 3: Generate the JSON string
  //
  root.printTo(Serial);
  Serial.println("");
  
}`

Sketch uses 77,104 bytes (69%) of program storage space. Maximum is 110,592 bytes.
Global variables use 6,080 bytes of dynamic memory.

My question is that Is it normal that this sketch takes about 75kB of flash space ?

Normally basic blink takes only around 12kB of flash space.

Sketch uses 13,760 bytes (12%) of program storage space. Maximum is 110,592 bytes.
Global variables use 2,816 bytes of dynamic memory.

maichaell, 2016-11-29:
Output of Arduino build:

`

C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\hardware -hardware C:\Users\michael\AppData\Local\Arduino15\packages -hardware C:\Users\michael\Documents\Arduino\hardware -tools C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\tools-builder -tools C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\hardware\tools\avr -tools C:\Users\michael\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\libraries -libraries C:\Users\michael\Documents\Arduino\libraries -fqbn=Arduino_STM32:STM32F1:mapleMini:bootloader_version=original -vid-pid=0X1EAF_0X0004 -ide-version=10611 -build-path C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp -warnings=more -prefs=build.warn_data_percentage=75 -verbose C:\Users\michael\Documents\Arduino\arduino_json_EEPROM\arduino_json_EEPROM.ino
C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\arduino-builder -compile -logger=machine -hardware C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\hardware -hardware C:\Users\michael\AppData\Local\Arduino15\packages -hardware C:\Users\michael\Documents\Arduino\hardware -tools C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\tools-builder -tools C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\hardware\tools\avr -tools C:\Users\michael\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\michael\Downloads\arduino-1.6.11-windows\arduino-1.6.11\libraries -libraries C:\Users\michael\Documents\Arduino\libraries -fqbn=Arduino_STM32:STM32F1:mapleMini:bootloader_version=original -vid-pid=0X1EAF_0X0004 -ide-version=10611 -build-path C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp -warnings=more -prefs=build.warn_data_percentage=75 -verbose C:\Users\michael\Documents\Arduino\arduino_json_EEPROM\arduino_json_EEPROM.ino
Using board 'mapleMini' from platform in folder: C:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1
Using core 'maple' from platform in folder: C:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1
Detecting libraries used...
"C:\Users\michael\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_maple_mini -DVECT_TAB_ADDR=0x8005000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10611 -DARDUINO_MAPLE_MINI -DARDUINO_ARCH_STM32F1   -DMCU_STM32F103CB -DSERIAL_USB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -DSERIAL_USB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/include" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/stm32f1/include"                                                           "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/usb/stm32f1" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/usb/usb_lib" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\maple_mini" "C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp\sketch\arduino_json_EEPROM.ino.cpp" -o "nul"
"C:\Users\michael\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_maple_mini -DVECT_TAB_ADDR=0x8005000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10611 -DARDUINO_MAPLE_MINI -DARDUINO_ARCH_STM32F1   -DMCU_STM32F103CB -DSERIAL_USB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -DSERIAL_USB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/include" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/stm32f1/include"                                                           "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/usb/stm32f1" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/usb/usb_lib" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple" "-IC:\Users\michael\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\maple_mini" "-IC:\Users\michael\Documents\Arduino\libraries\ArduinoJson" "C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp\sketch\arduino_json_EEPROM.ino.cpp" -o "nul"
"C:\Users\michael\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE  -ffunction-sections -fdata-sections -nostdlib --param max-inline-ins [... cut ...]
"C:\Users\michael\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-objcopy" -O binary  "C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp/arduino_json_EEPROM.ino.elf" "C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp/arduino_json_EEPROM.ino.bin"
Using library ArduinoJson at version 5.7.2 in folder: C:\Users\michael\Documents\Arduino\libraries\ArduinoJson 

Sketch uses 77,104 bytes (69%) of program storage space. Maximum is 110,592 bytes.
Global variables use 6,080 bytes of dynamic memory.
`

bblanchon, 2016-11-29:
No, this is not normal.

ArduinoJson uses a lot of template code and expects that the compiler will inline it.
For some reason, it's not happening here.
Maybe this is related to `max-inline-insns-single=500` on the GCC command line by I'm not sure.

Anyway, we need to know what exactly is eating up all the space.
Can you do something similar to what I did in [this blog post](http://blog.benoitblanchon.fr/arduino-code-size-and-assembly/)?
If you cannot, that's OK, I'll have a look at it from home.

maichaell, 2016-11-29:
Sure, no problems :-)

Here: [disassemlbe_all.zip](https://github.com/bblanchon/ArduinoJson/files/618807/disassemlbe_all.zip)

cmdLine: `arm-none-eabi-objdump.exe -d -C C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp\arduino_json_EEPROM.ino.elf > c:\disassemlbe_all.txt`

maichaell, 2016-11-29:
Would be nice to have an example of how to use EEPROM library and save / read JSON strings from. Are there any ? In STM32Duino EEPROM library is "emulated" and in realityit uses flash - so there's quite nice amount of extra space for JSON :-)

bblanchon, 2016-11-29:
It seems that ArduinoJson only takes 2% of the space.
Can you run:

    arm-none-eabi-nm.exe --size-sort -C -r C:\Users\michael\AppData\Local\Temp\build6e0206c4052a2b088e47e64ac9275fbc.tmp\arduino_json_EEPROM.ino.elf

(Please open a new issue for your question related to EEPROM)

maichaell, 2016-11-29:
Yes, this seems strange.

Here you go:

[nm.txt](https://github.com/bblanchon/ArduinoJson/files/619066/nm.txt)

maichaell, 2016-11-29:
zipping here the whole Arduino build folder:

[build6e0206c4052a2b088e47e64ac9275fbc.tmp.zip](https://github.com/bblanchon/ArduinoJson/files/619079/build6e0206c4052a2b088e47e64ac9275fbc.tmp.zip)

bblanchon, 2016-11-29:
If you grep through `nm.txt`, you'll see that ArduinoJson is just 1104 bytes.

    grep ArduinoJson nm.txt | cut -d ' ' -f1 | xargs -I {} -n1 printf "%d\n" '0x{}' | awk '{ sum += $1 } END { print "ArduinoJson takes " sum " bytes" }'

Since the blink program is way smaller (but still big admittedly), I think that most of the size is due to the `Serial` class and its dependencies.

maichaell, 2016-11-29:
Yes, you're right. ArduinoJson does not occupy much. Blink takes only 12kb and when I add ArduinoJson example code the size rises to 76kB. Strange.  Some other lib must be pulling a lot of dependencies or is the Arduino flash size calculation wrong? 

Anyway, I'll open a discussion about this in STM32Duino forum since your code is fully OK. Thanks for you time!

bblanchon, 2016-11-29:
Try to add this line to the blink example:

    Serial.println("You know I'm fat, I'm fat, you know it");

I'm pretty sure the size will increase a lot.

maichaell, 2016-11-29:
```
#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  Serial.println("You know I'm fat, I'm fat, you know it");
 
  StaticJsonBuffer<200> jsonBuffer;
  
  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
 
  //JsonObject& root = jsonBuffer.parseObject(json);

}

void loop() { }

```
It produces: 

```
Sketch uses 12,720 bytes (11%) of program storage space. Maximum is 110,592 bytes.
Global variables use 2,816 bytes of dynamic memory.
```

When I uncomment this:  //JsonObject& root = jsonBuffer.parseObject(json);

The size increases:
```

Sketch uses 73,840 bytes (66%) of program storage space. Maximum is 110,592 bytes.
Global variables use 6,080 bytes of dynamic memory.
```

bblanchon, 2016-11-29:
That is very interesting 👍 

Can you send the `nm` output before and after, so we can see what is actually added?

maichaell, 2016-11-29:
Here we go:

[nm.zip](https://github.com/bblanchon/ArduinoJson/files/620039/nm.zip)

bblanchon, 2016-11-29:
I manage to cut the size in have by adding `-fno-threadsafe-statics` in the compiler command line.

File:

    %LOCALAPPDATA%\Arduino15\packages\stm32duino\hardware\STM32F1\1.0.10\platform.txt

Line 25:

     compiler.cpp.flags=... -fno-threadsafe-statics

This tells the compiler to disable thread safety on static variable initialization, which is perfectly safe on a single threaded application.

maichaell, 2016-11-30:
Thanks. I tried your suggestion and it did cut the size in half.

But the solution seems to be this:

add: 

--specs=nano.specs

at the very end of line in Documents\Arduino\hardware\Arduino_STM32\STM32F1\platform.txt on the line that starts with "recipe.c.combine.pattern=...."

as suggested here:

http://www.stm32duino.com/viewtopic.php?f=28&t=1596&p=20642#p20627

With this the size is what is should be i.e. 1kB larger than basic 'blink' program.

maichaell, 2016-12-05:
I've done some coding with this library and just wanted to say it works very well and is very flexible. Very good API. The way you use templates is something I've never seen before .. and take that as a compliment :-) On my behalf this can be closed when you see necessary.

bblanchon, 2016-12-07:
Thanks for the kind words @maichaell, and thank you for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-12-07 by bblanchon