bblanchon/ArduinoJson #396: Including ArduinoJson from subdir of Arduino project/sketch

Humphreybas, 2016-12-02:
I don't expect this to be 'solved' (certainly since it's not actually a problem of the ArduinoJson software), but maybe someone has some insights or tips on this:

For git, portability and easy deployment of my Arduino project, I would like to keep the ArduinoJson library inside the project/sketch folder. With the Arduino IDE this *was* ridiculous hard to do, they wanted you to use the global library folder. However now that I am working with version 1.6.13 of the IDE it's a bit easier. You can actually include a relative path:
`#include "ArduinoJson/ArduinoJson.h"`

Unfortunately, this does not work for ArduinoJson because JsonParser.ipp cannot be found, and as I learned in a [arduino forum thread](http://forum.arduino.cc/index.php?topic=400477.msg2755702#msg2755702) this is because the Arduino IDE does not copy *.ipp files to the temp directory in which it is building.
I tried renaming the ipp files to cpp but that gave all sorts of errors. Is there by any change someone who has a tip how to solve this?

bblanchon, 2016-12-02:
I wasn't aware of this problem, I'll try to remove the `.ipp` in a future release.

As a temporary workaround, I suggest that you rename `*.ipp` to `*Impl.hpp`, and of course update the `#include`s in `include/ArduinoJson.hpp`.

Humphreybas, 2016-12-02:
That is great man, works brilliant, thanks!

aacton, 2016-12-02:
@Humphreybas I've encountered the same issue: I want to keep a copy of Arduino libraries inside my project git repo. I do this with git subtree, but as you point out, Arduino wants the library in the global folder. On linux, I simply go into the ~/Arduino/libraries folder and symlink to the directory in my repo that has the library code, in this case would be ArduinoJson. It works very well for me. If you're working with Windows, I think the equivalent operation is mklink.

Humphreybas, 2016-12-03:
That is a nice solution as well. However personally I prefer the relative import (which is now working without any problems). The symbolic link solution is more of a hassle, you have to instruct the user to make a symlink, not easy for everyone. And when you move/rename any folder in the path to the library you have to recreate the link.

bblanchon, 2016-12-03:
The files have been renamed in the master branch.

closed 2016-12-03 by bblanchon