bblanchon/ArduinoJson #397: Json HTTP Client - unexpecteted Datas

Chris685, 2016-12-13:
If i use a longer Data string like:

const char* server = "jsonplaceholder.typicode.com";  // server's address
const char* resource = "/users"; // http resource

I got an "160d" before my Datas - i have no idea why. 

I am using the Arduino DUE with an WIFI Shield. 

Thanks

Serial Output of Content (shorted):

"
160d
[
  {
    "id": 1,
    "name": "Leanne Graham",
    "username": "Bret",
    "email": "Sincere@april.biz",
    "address": {
      "street": "Kulas Light",
      "suite": "Apt. 556",
      "city": "Gwenborough",
      "zipcode": "92998-3874",
      "geo": {
        "lat": "-37.3159",
        "lng": "81.1496"
      }
    },
    "phone": "1-770-736-8031 x56442",
    "website": "hildegard.org",
    "company": {
      "name": "Romaguera-Crona",
      "catchPhrase": "Multi-layered client-server neural-net",

bblanchon, 2016-12-14:
Hi Chris,

This is [HTTP Chunked Transfer Encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).
The server is sending a response in multiple parts, except that in this particular case, it's just one.

`160d` is the number of bytes in the first chunk, you're supposed to parse that hexadecimal number and read the 5645 bytes that follow. Then, you'll receive the size of the next chunk, which is `0` in this case.

As far as I know, [it's not possible to disable this feature in HTTP 1.1](http://stackoverflow.com/a/23774393/1164966).
To disable this feature, you need to change the GET request to use HTTP 1.0

Chris, please tell me if this works.
If so I'll update the `JsonHttpClient` example accordingly.
You can make a Pull Request if you prefer, I'll be glad to merge it.

Chris685, 2016-12-14:
Hello Benoît,

this is working. Thank you very much. Now i have to read in to the topic JSON Decoding :).

Below the Code is useing now:

``
// Send the HTTP GET request to the server
bool sendRequest(const char* host, const char* resource) {
  Serial.print("GET ");
  Serial.println(resource);
  client.print("GET ");
  client.print(resource);
  client.println(" HTTP/1.0");
  client.print("Host: ");
  client.println(server);
  client.println("Connection: close");
  client.println();

  return true;
}
``

Chris685, 2016-12-14:
One more question to my specific Datas. 

Is it possible to get on this informations? For example "Temperatur Sensor 2"

Thanks

`{
	"language" : "de",
	"headers" : [
	{
		"id" : "00_0010_7751_0100",
		"description" : "VBus 0: DeDietrich Diemasol C v2007",
		"channel" : 0,
		"destination_address" : 16,
		"source_address" : 30545,
		"protocol_version" : 16,
		"command" : 256,
		"info" : 0,
		"destination_name" : "DFA",
		"source_name" : "DeDietrich Diemasol C v2007",
		"fields" : [
		{
			"id" : "000_2_0",
			"name" : "Temperatur Sensor 1",
			"unit" : " \u00B0C",
			"unit_code" : "DegreesCelsius"
		},
		{
			"id" : "002_2_0",
			"name" : "Temperatur Sensor 2",
			"unit" : " \u00B0C",
			"unit_code" : "DegreesCelsius"
		},`

closed 2016-12-14 by Chris685