bblanchon/ArduinoJson #398: Example of converting String for use with ArduinoJson?

Humancell, 2016-12-13:
I have dug and dug and looked everywhere ... and I'm sure there HAS to be a clear example of how to properly convert a String to use with ArduinoJson ... but I just can not find one.

EVERYONE seems to want to just say "Don't use Strings!" ... but we HAVE TO as we already have a ton of code that can not be rewritten ... and we have the spare memory.  So we THOUGHT we could simply convert the String to a char array using the standard functions ... but it simply refuses to work ... and we know we are doing something wrong.

```
  String responseBodyString = "{"
      "\"status\":\"success\","
      "\"version\":\"1.0\","
      "\"timeEpoch\":1480823123,"
      "\"errorURL\":\"http://ws.website.net/errors\","
      "\"actions\": ["
        "{"
          "\"action\":\"updateProperty\","
          "\"property\":\"some.property.1\","
          "\"value\":\"some value 1\""
        "},"
        "{"
          "\"action\":\"updateProperty\","
          "\"property\":\"some.property.2\","
          "\"value\":\"some value 2\""
        "},"
        "{"
          "\"action\":\"updateProperty\","
          "\"property\":\"some.property.3\","
          "\"value\":\"some value 3\""
        "}"
      "]"
    "}";

  char responseBody[] = "{"
      "\"status\":\"success\","
      "\"version\":\"1.0\","
      "\"timeEpoch\":1480823123,"
      "\"errorURL\":\"http://ws.wovyn.net/errors\","
      "\"actions\": ["
        "{"
          "\"action\":\"updateProperty\","
          "\"property\":\"some.property.1\","
          "\"value\":\"some value 1\""
        "},"
        "{"
          "\"action\":\"updateProperty\","
          "\"property\":\"some.property.2\","
          "\"value\":\"some value 2\""
        "},"
        "{"
          "\"action\":\"updateProperty\","
          "\"property\":\"some.property.3\","
          "\"value\":\"some value 3\""
        "}"
      "]"
    "}";

  int lengthString = responseBodyString.length();
  Serial.print(F("JSON String response length: "));  // always one char shorter than the char[]
  Serial.println(lengthString);

  int length = sizeof(responseBody);
  Serial.print(F("JSON response length: "));  // always one char longer than the string?
  Serial.println(length);
  
  // Allocate a buffer to store contents of the file.
  // std::unique_ptr<char[]> buf(new char[length]);
  
  // We don't use String here because ArduinoJson library requires the input
  // buffer to be mutable. If you don't use ArduinoJson, you may as well
  // use configFile.readString instead.
  //  configFile.readBytes(buf.get(), size);

  // attempt to convert the String to char[]
  char  buf[lengthString + 1];
  responseBodyString.toCharArray(buf, lengthString);
  buf[lengthString] = 0;  // zero terminate?  This was a guess ...
  
  StaticJsonBuffer<500> jsonStringBuffer;
  JsonObject& jsonString = jsonStringBuffer.parseObject(responseBodyString);

  if (!jsonString.success()) {
    Serial.println(F("Failed to parse responseBodyString"));  // ALWAYS FAILS!   ;-(
    return false;
  }

  StaticJsonBuffer<500> jsonBuffer;
  JsonObject& json = jsonBuffer.parseObject(responseBody);

  if (!json.success()) {
    Serial.println(F("Failed to parse responseBody"));  // ALWAYS WORKS!  ;-(
    return false;
  }

```

I know there must be something simple we are missing, and not understanding about doing this conversion.  We don't see any other error, except that the parse fails.

Thank you in advance ... this is driving us nuts!

Humancell, 2016-12-14:
Ok ... there are TWO bugs in the code above.  One is a stupid error during debugging, and the other is related to the buffer size.

In this code snippet - updated from the code above - the attempt to parseObject(responseBodyString); should have been to parseObject(buf); ... the stupid error.

```
  char  buf[lengthString + 1];
  responseBodyString.toCharArray(buf, lengthString + 1);  // THIS had to be enlarged to == buf size
  
  int bufLength = sizeof(buf);
  Serial.print(F("JSON buf response length: "));
  Serial.println(bufLength);

  StaticJsonBuffer<500> jsonStringBuffer;
  JsonObject& jsonString = jsonStringBuffer.parseObject(buf);

```

What fixed the problem was changing the parameter being passed to the .toCharArray() to be .toCharArray(buf, lengthString + 1)  What we seemed to miss is that the buffer has to be one char larger than the String length, and the .toCharArray() ALSO requires this enlarged buffer size.

bblanchon, 2016-12-14:
Hi Scott,

`parseObject()` can take a `String` as input; but in that case, it will have to make a copy of if in the `JsonBuffer`, so you need to increase the size of the buffer.

As an alternative, you can ignore the const-ness of the string so as to avoid the duplication.

```
JsonObject& root = jsonBuffer.parseObject(static_const<char*>(myString.c_str()));
```

This will modify the string in place.
It is safe, as long as the `String` stays in memory.

closed 2016-12-21 by bblanchon

Humancell, 2016-12-21:
Sorry ... I failed to thank you for the follow-up!  It's been working great, and I better understand the documentation and example you provided.  Great library, and it's working awesome.

bblanchon, 2016-12-22:
You're welcome Scott, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: