bblanchon/ArduinoJson #400: How to decode arrays in array?

kiralikbeyin, 2016-12-21:
String tester = "[[\"🕐\",\"Pt\",\"Sa\",\"Çr\",\"Pe\",\"Cm\",\"Ct\",\"Pz\"],[\"00-01\",\"21\",\"21\",\"21\",\"21\",\"21\",\"21\",\"21\"],[\"01-02\",\"21\",\"21\",\"21\",\"21\",\"21\",\"21\",\"21\"]]";

Couldn't find an example, is it possible?

bblanchon, 2016-12-21:
```c++
JsonArray& root = jsonBuffer.parseArray(tester);
const char* value = root[0][0];
```

kiralikbeyin, 2016-12-21:
@bblanchon Thank you but i changed it because i get empty responses (Half of this son data works)

Tried 1024-2048-4096 Staticbuffer but didn't work. (Nodemcu 1.0)
Latest ArduinoJson lib installed

[{"Saat":"00-01","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"01-02","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"02-03","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"03-04","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"04-05","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"05-06","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"06-07","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"07-08","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"08-09","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"09-10","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"10-11","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"11-12","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"12-13","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"13-14","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"14-15","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"15-16","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"16-17","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"17-18","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"18-19","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"19-20","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"20-21","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"21-22","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"22-23","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"},{"Saat":"23-00","Pt":"21","Sa":"21","Çr":"21","Pe":"21","Cm":"21","Ct":"21","Pz":"21"}]

bblanchon, 2016-12-22:
The [JsonBuffer size calculator](https://bblanchon.github.io/ArduinoJson/) says that you need 3560 bytes to store the object tree, so a `JsonBuffer` of 4096 bytes is appropriate.

But it's **way too big for the stack** of an ESP8266; you need to move that memory to the heap.

To do this, just replace:

    StaticJsonBuffer<4096> jsonBufferhaft;

by:

    DynamicJsonBuffer jsonBufferhaft(4096);

Also, I see that the `haft` array is using a significant part of the stack, you may need to move it to the heap too.

By the way, I see that you switch from `String` to `char[]` which is a good idea, since it avoids the duplication of the JSON input in the `JsonBuffer` :+1:

One last thing, you could get a significant performance boost by using iterators:

```c++
for (JsonArray& arr : jsonhaft) {
  haft[i][1] = arr["Pt"];
  haft[i][2] = arr["Sa"];
  haft[i][3] = arr["Çr"];
  haft[i][4] = arr["Pe"];
  haft[i][5] = arr["Cm"];
  haft[i][6] = arr["Ct"];
  haft[i][7] = arr["Pz"];
  i++;
}
```

kiralikbeyin, 2016-12-24:
Thank you it works fine.

I didnt understand this because of [i]
```
for (JsonArray& arr : jsonhaft) {
  haft[i][1] = arr["Pt"];
  haft[i][2] = arr["Sa"];
  haft[i][3] = arr["Çr"];
  haft[i][4] = arr["Pe"];
  haft[i][5] = arr["Cm"];
  haft[i][6] = arr["Ct"];
  haft[i][7] = arr["Pz"];
  i++;
}
```

bblanchon, 2016-12-24:
You're welcome @kiralikbeyin, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-12-24 by bblanchon