bblanchon/ArduinoJson #401: JSON Strings with quotes ...

Humancell, 2016-12-23:
I'm slowly getting the hang of the library, and I'm parsing ok, but now I've hit an issue with comparing/checking values of the properties ... specifically Strings.

Using your example, if I have this JSON: 

```
 char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
```

I can then successfully parse it.

Now, I want to check a value ... and this continues to fail:

```
  Serial.print("sensor: ");
  if (json.containsKey("sensor"))
  {
    json["sensor"].printTo(Serial);  // NOTE: This prints "gps" with the quotes!
    Serial.println();

    if (json["sensor"]) == "gps") {  // NOTE: Always fails!  Unless changes to "\"gps\""
      /*
       *   How can I compare/get the value WITHOUT the quotes?
       *    - I don't want to have to always put escaped quotes in all comparisons?
       */
    }

```

How can I do a comparison of a string, but drop the quotes?  Or reference the JSON string without the quotes?

bblanchon, 2016-12-23:
`printTo()` is meant to serialize a value to its JSON form, so the quotes are expected.

To get the string out of `json["sensor"]`, there are three syntaxes:

    const char* sensor = json["sensor"];
    const char* sensor = json["sensor"].as<char*>();
    const char* sensor = json.get<char*>("sensor");

To compare `char*` strings, you need to call `strcmp()` like this:

    if (strcmp(sensor, "gps") == 0) {

    }

Now, I agree that there is something wrong because this should have worked too:

    if (json["sensor"] == "gps") {
 
    }

This looks like a bug, let me see if I can fix it...

Humancell, 2016-12-23:
Thank you for your quick reply!

Just so that you know, I had to use the following syntax to get it to work:

```
    if (strcmp(json["sensor"], "\"gps\"") == 0) {
      Serial.println(F("sensor is a gps"));
      /*
       *   Again here if I don't escape the quotes it fails ...
       *    - I'll assume this is because I did not properly de-reference it
       */
      return true;
    }
```

I really like your fix to support:

```
if (json["sensor"] == F("gps")) {

}
```

I'll grab the latest version and test later today!

bblanchon, 2016-12-24:
Hi Scott,
This `strcmp(json["sensor"], "gps")` is really bothering me.
I'm pretty sure it works as expected.
See [this passing unit test](https://github.com/bblanchon/ArduinoJson/blob/0d8204bffa13610dc3d1bca7c433e273fe606028/test/Issue401.cpp)
Can you please check once again?

Humancell, 2016-12-31:
Ok ... my bad.

If you look above, I was actually using the wrong comparison to detect when the strings don't "match".  I was looking for the "fail" condition, and forgot that "0" indicates a match.

`if (strcmp(json["sensor"], "gps") == 0) {
  // this is where we fail
`
I should have been using:

`if (strcmp(json["sensor"], "gps") != 0) {
  // this is where we fail
`

Your code is correct.  :-)

bblanchon, 2016-12-31:
Good. Thanks.

closed 2016-12-31 by bblanchon

Humancell, 2016-12-31:
BTW ... when will you create a update/release that includes the fix:

```
if (json["sensor"] == F("gps")) {

}
```

This will read much better ... :-)

Humancell, 2017-01-04:
Also ... will `!=` work also?

bblanchon, 2017-01-05:
Yes `!=` will too.
But `<`, `>`, `<=`, `>=` won't.

Humancell, 2017-01-05:
Excellent!  And I got the update ... I'll continue testing.

Thanks again!